olofk/fusesoc #222: Vivado synthesis flow fails with KeyError: 'hw_device'

lerwys, 2018-08-06:
On running the following command:

`fusesoc run --target=synth --tool=vivado picorv32_demo`

fusesoc fails after bitstream generation with the message:

```
Writing bitstream ./picorv32_demo_0.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Project 1-120] WebTalk data collection is mandatory when using a WebPACK part without a full Vivado license. To see the specific WebTalk data collected for your design, open the usage_statistics_webtalk.html or usage_statistics_webtalk.xml file in the implementation directory.
INFO: [Common 17-186] '/home/lerwys/Repos/fpga-flow/build/picorv32_demo_0/synth-vivado/usage_statistics_webtalk.xml' has been successfully sent to Xilinx on Mon Aug  6 09:23:20 2018. For additional details about this file, please refer to the WebTalk help file at /opt/Xilinx/Vivado/2018.2/doc/webtalk_introduction.html.
INFO: [Common 17-83] Releasing license: Implementation
12 Infos, 1 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:00:08 ; elapsed = 00:02:20 . Memory (MB): peak = 2324.020 ; gain = 441.668 ; free physical = 3212 ; free virtual = 21288
INFO: [Common 17-206] Exiting Vivado at Mon Aug  6 09:23:20 2018...
Traceback (most recent call last):
  File "/usr/local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc', 'console_scripts', 'fusesoc')()
  File "/home/lerwys/Repos/fusesoc/fusesoc/main.py", line 531, in main
    args.func(cm, args)
  File "/home/lerwys/Repos/fusesoc/fusesoc/main.py", line 235, in run
    flags, args.system, args.backendargs)
  File "/home/lerwys/Repos/fusesoc/fusesoc/main.py", line 316, in run_backend
    backend.run(backendargs)
  File "/home/lerwys/Repos/fusesoc/edalize/vivado.py", line 108, in run
    self._write_program_tcl_file(tcl_file_name)
  File "/home/lerwys/Repos/fusesoc/edalize/vivado.py", line 115, in _write_program_tcl_file
    template_vars['hw_device'] = self.tool_options['hw_device']
KeyError: 'hw_device'
```

I'm running the following fusesoc version:

```
lerwys@lerwysPC:~/Repos/fpga-flow$ fusesoc --version
1.8.4.dev5+gfd6df63
```

From: [https://github.com/lerwys/fusesoc](https://github.com/lerwys/fusesoc)

Core file:

```
lerwys@lerwysPC:~/Repos/fpga-flow$ cat syn/picorv32_demo/picorv32_demo.core 
CAPI=2:
name : ::picorv32_demo:0

filesets:
  core_files:
    files:
      - ../../top/picorv32_demo/wb_intercon.v
      - ../../top/picorv32_demo/wb_intercon.vh:         {is_include_file : true}
      - ../../top/picorv32_demo/picorv32_demo_system.v
    file_type: verilogSource
    depend:
      - ">=wb_intercon-1.0"
      - ">=wb_ram-1.0"
      - picorv32
      - wb_leds

  top_files:
    files:
      - ../../platform/xilinx/series7/sys_clk.v
      - ../../top/picorv32_demo/picorv32_demo_top.v
    file_type : verilogSource

  constraints :
    files:
      - ../../top/picorv32_demo/picorv32_demo.xdc:      {file_type: xdc}

targets:
  synth: &synth
    default_tool : vivado
    filesets   :
      - core_files
      - top_files
      - constraints
    tools:
      vivado:
        part: xc7a35tcpg236-1
        hw_device: xc7a35t_0
    toplevel: [picorv32_demo_top]

  default:
    <<: *synth

parameters:
  firmware_file:
    datatype    : file
    description : Initial boot ROM contents (in Verilog hex format)
    paramtype   : vlogparam
    scope       : private

#provider:
#  name    : github
#  user    : lerwys
#  repo    : fpga-flow
# version : v0
```

Project repository: [https://github.com/lerwys/fpga-flow](https://github.com/lerwys/fpga-flow)

Applying the following patch seems to solve the issue, but I wanted to ask if I'm missing something:
https://github.com/lerwys/fusesoc/commit/7adbbd0d33a2533ca6b4237a03ae460eea5affe3#diff-b2231054479c23c01c572217f89786b8

Thanks!

Lucas

olofk, 2018-08-06:
Hi Lucas,

Your fix is correct. At the same time I'm hoping it will be unnecessary. The thing is that `hw_device` is only used when FuseSoC tries to program the FPGA after the bitstream has been generated. If you look at https://github.com/olofk/fusesoc/pull/210 I have proposed an alternative method of programming devices that won't need `hw_device` at all, but I haven't had any opportunity to test it on real hardware.

Another thing that you might not be aware of is that `fusesoc run...` will create the bitstream and try to program the device through Vivado, while `fusesoc build...` will stop after bitstream generation. So if all you need is to create the bitstream, use `fusesoc build` for now until the device programming side has been fixed properly

lerwys, 2018-08-06:
Hi Olof,

Sure, lazy me didn't go through all of the fusesoc options. =] 

I had just learned that I could run `fusesoc build` when I sent the issue. I will use that as all I need is the bitstream.

About #210. Sounds a good idea to get rid of the hw_devices, as it is a bit redundant anyway.

Thanks!

olofk, 2019-04-15:
This should be fixed now with https://github.com/olofk/edalize/commit/3f087d27323765ba656790d8d0cef8c433c70f2f

closed 2019-04-15 by olofk