olofk/fusesoc #233: CAPI2: --build without --setup gives Python exception

imphil, 2018-11-12:
In CAPI1 we had `fusesoc sim --build-only` to only configure and build a simulation, but not to run it.
In CAPI2, the both configure/setup (confusingly with different names in source and command line) are not implied when selecting the build phase, leading to something like `fusesoc run --target=sim --tool=verilator --setup --build` to achieve similar outcome.

If only specifying `--build` without `--setup` we get this:

```
Traceback (most recent call last):
  File "/home/no56hud/.local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc', 'console_scripts', 'fusesoc')()
  File "/data/home/no56hud/src/fusesoc/fusesoc/main.py", line 570, in main
    args.func(cm, args)
  File "/data/home/no56hud/src/fusesoc/fusesoc/main.py", line 259, in run
    flags, args.system, args.backendargs)
  File "/data/home/no56hud/src/fusesoc/fusesoc/main.py", line 311, in run_backend
    backend = get_edatool(tool)(eda_api=edalizer.edalize,
UnboundLocalError: local variable 'edalizer' referenced before assignment
```

Two options to solve this:
1. Give a proper error message when specifying only the build phase without the configure/sim phase.
2. Imply all previous phases when selecting a phase, i.e. imply `--setup` when running `--build`, and imply both options when specifying `--run`.

I'd prefer option 2.

olofk, 2018-11-30:
I think a variation of option 2 would be best, because there are some things to take into consideration.

For the run phase we likely don't want this, as we want to be able to run e.g. several simulations with different parameters without rebuilding every time. But this implies we would need to detect if the build phase has been run.

For the build phase it's less of a problem, since the setup phase tends to be pretty quick. But perhaps there are still reasons to avoid running setup again (I agree that the naming of the stages are terribly confusing. Sorry about that :/). So I think once again we could use detection instead.

So, how to do detection? We could drop in indicator files, but really, I think it's easier to just check for the presence of a Makefile to see if setup was completed. It looks like only the riviera-pro backend doesn't create a makefile during configure so we could just update that backend

For the build phase it's even easier. In most cases, the run stage only launches `make run` with some runtime parameters. If the makefile is setup already, it should build the project automatically if missing

olofk, 2019-04-12:
Sorry for taking so long. This should hopefully be fixed now with ae9cdc3
Otherwise, shout and reopen. Thanks for reporting

closed 2019-04-12 by olofk