olofk/fusesoc #235: Choose between implementations of an interface

imphil, 2018-12-13:
This is essentially a question/starting point for discussions about a use case which probably has happened before, and I'd like to get @olofk's thoughts on the topic. (I'm very sorry for the long text, and it's probably restating much stuff that is mostly clear, but better be on the safe side ...)

**Scenario**
(This is rather specific, you could do it with similar constructs like defines as well.)
- I have a SystemVerilog package which defines common parameters for a bus, like data and address width. Something like that:

  ```sv
  // modules/vendor/bus/rtl/bus_conf_pkg_default.sv
  package bus_conf_pkg;
    localparam BUS_AW = 32;
    localparam BUS_DW = 32;
  endpackage;
   ```

- Code in my bus module uses the `bus_conf_pkg` package as a source of global configuration, e.g.

  ```sv
  // modules/vendor/bus/rtl/bus_top.sv
  module bus_top(
    input [bus_conf_pkg::BUS_AW-1:0] addr
  );
  // ...
  endmodule
   ```
- I want to create a reusable module with a FuseSoC description. This means here: Give the user of my module the chance to configure the bus project wide by providing their own implementation of `bus_conf_pkg`. In practice this means the user should be required to include a FuseSoC core which provides the `bus_conf_pkg` package, but I don't want to force the user to take a specific one.

**Current FuseSoC solution**
- Create a core file which provides the `bus_conf_pkg` package.
  ```yaml
  CAPI=2:
  # modules/vendor/bus/config.core
  name: vendor:bus:config
  description: "The bus configuration"

  filesets:
    files_rtl:
      files:
        - rtl/bus_conf_pkg_default.sv
      file_type: systemVerilogSource
  ```

- Create another core file for the implementation which depends on a core providing the `bus_conf_pkg` package.

  ```yaml
  CAPI=2:
  # modules/vendor/bus/implementation.core
  name: vendor:bus:implementation
  description: "The bus implementation"

  filesets:
    files_rtl:
      files:
        - rtl/bus_top.sv
      file_type: systemVerilogSource

      depend:
        - vendor:bus:config
  ```

- I ship the module with both core files, and an implementation using the core adds a `depend: [ vendor:bus:implementation ]` to its core file and it can happily use the bus with the default values given in `bus_conf_pkg_default.sv`.

The challenge starts when one wants to override the default configuration, i.e. provide an own implementation of `bus_conf_pkg`. I see the following options currently:

1. Override the `vendor:bus:config` core with a customized core implementation. Since cores are referenced by name, this means I can create another core file equal to the `modules/vendor/bus/rtl/config.core` file, let's call it `system/bus_config.core`. For this file to override the `modules/vendor/bus/rtl/config.core` file it must be placed in the CORES_ROOT search path (before or after, I always forget) the other file. 

    This approach works with fusesoc today, but it's error prone: if the search path is messed up, the design ends up synthesized just fine, but with the wrong configuration values (as `bus_conf_pkg_default.sv` is picked up instead of the customized copy). Messing up the search path is easy, since this piece of configuration depends on the user's build environment.

2. Do not declare the dependency on the `bus_config_pkg` in FuseSoC, i.e. omit the `depend: [ vendor:bus:config ]` stance from the `modules/vendor/bus/implementation.core` core file. Users then explicitly need to include any core file which provides a package named `bus_config_pkg`, otherwise they get compile errors from their synthesis tool.

    Again, that works today and it's fairly safe to do, however it circumvents the dependency resolution in FuseSoC and relies on the backend tool instead. That's not ideal either.

Since none of the solutions that I came up with are ideal, I thought about a better solution.

**A better solution**

Goals:
- All dependencies should be stated explicitly in the core files.
- Do not rely on the ordering of directories in the search path to determine which cores are used.

The idea essentially is to introduce a `provides` option in the core file, which specifies an "interface" a core implements. Other cores which require an implementation of this interface specify the interface name in their `depend` line, which looks exactly as before.

```yaml
CAPI=2:
name: vendor:bus:implementation
description: "The bus implementation"

filesets:
  files_rtl:
    files:
      - rtl/bus_top.sv
    file_type: systemVerilogSource

    depend:
      - vendor:bus:config
```

```yaml
CAPI=2:
name: vendor:bus:config_default
provides: vendor:bus:config # new syntax here
description: "The default bus configuration"

filesets:
  files_rtl:
    files:
      - rtl/bus_conf_pkg.sv
    file_type: systemVerilogSource
```

```yaml
CAPI=2:
name: othervendor:project_x:busconfig
provides: vendor:bus:config # new syntax here
description: "The bus configuration specific to project_x"

filesets:
  files_rtl:
    files:
      - rtl/bus_conf_pkg.sv
    file_type: systemVerilogSource
```

Now to use the bus, you need to explicitly choose which "provider" (implementation) of `vendor:bus:config_default` you want to include:

```yaml
CAPI=2:
name: othervendor:project_x:top
description: "The toplevel core file of project_x"

depend:
- vendor:bus:implementation
- othervendor:project_x:busconfig
```

Final notes:
- To keep things simple I've specified `provides` as a string. It could also be an list if a single core implements multiple interfaces. That's probably overkill, however.
- `provides` is used as a keyword in RPM, for fusesoc `provide` is more in line with `depend` (which I always mis-spell as `depends` ...)

So, after reading this whole essay @olofk, are you going to tell me that something like this is already implemented and waiting to be used? :-)

olofk, 2018-12-14:
Thank you for the careful analysis. More words is definitely better than less :)

I have thought about a `provide` options before, but have decided to not implement it because it adds some complexity and there haven't been any real use cases since I haven't been able to come up with any two cores that could act as drop-in replacements for each other.

What you present is a slightly different use case, and I think we should revisit a `provide` option. (The underlying package manager already supports the concept which probably helps too). But I also want to think about alternative solutions.

Come to think about it, this is somewhat similar to the problem of switching in tool/vendor-specific RAM primitives, something that is still not super smooth in FuseSoC. To fix that problem, I plan to implement use flags. The support is basically there already, just not exposed on the command-line. Might be good to think if that could help here too

benreynwar, 2019-01-18:
This feels like a perfect use case for a generator.

The generator would take BUS_AW and BUS_DW as parameters and then create the SV configuration file.  Am I missing something?

PR #524 (2022-03-23): Support virtual packages in CAPI2
Packages can now provide more than one package: they provide itself (as
described by the package `name`), and a list of further (virtual)
packages, described by the `virtual` key in the root of the CAPI2 file.

Example:
```
CAPI=2:
name: ::impl2:0
virtual:
  - ::someinterface:0
```

This core provides a core named `::impl2:0` and a core named
`someinterface:0`. Other cores can now depend either `::impl2:0` or
`::someinterface` and always get the `::impl2:0` core.

The keyword `virtual` was chosen over alternatives, such as `provide` to
be distinct from other k [... cut]
commit 6b040284 (2022-03-23): Support virtual packages in CAPI2

Packages can now provide more than one package: they provide itself (as
described by the package `name`), and a list of further (virtual)
packages, described by the `virtual` key in the root of the CAPI2 file.

Example:
```
CAPI=2:
name: ::impl2:0
virtual:
  - ::someinterface:0
```

This core provides a core named `::impl2:0` and a core named
`someinterface:0`. Other cores can now depend either `::impl2:0` or
`::someinterface` and always get the `::impl2:0` core.

The keyword `virtual` was chosen over alternatives, such as `provide` to
be distinct from other  [... cut]
closed 2022-03-23 by imphil