olofk/fusesoc #239: Problem programming de0 nano project built with fusesoc

BryanOrabutt, 2019-02-11:
Im having an issue using the fusesoc pgm de0_nano command. I used `fusesoc build de0_nano --bootrom_file=~/.local/share/fusesoc/orpsoc-cores/systems/de0_nano/sw/spi_uimage_loader.vh --spi_flash_file=./vmlinux` and the project built with no errors.

However when I use `fusesoc pgm de0_nano` I get the following error messages:
`  File "/usr/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc==1.9', 'console_scripts', 'fusesoc')()
  File "/usr/lib/python3.7/site-packages/fusesoc/main.py", line 569, in main
    args.func(cm, args)
  File "/usr/lib/python3.7/site-packages/fusesoc/main.py", line 78, in pgm
    flags, args.system, args.backendargs)
  File "/usr/lib/python3.7/site-packages/fusesoc/main.py", line 310, in run_backend
    backend = get_edatool(tool)(eda_api=edalizer.edalize,
UnboundLocalError: local variable 'edalizer' referenced before assignment
`
I have tried using the `quartus_pgm --mode=jtag -o p\;de0_nano.sof` but I don't get any linux shell output over my FTDI cable connected to the UART. I also tried other elf files instead of the vmlinux but none of them seem to work. Any help is appreciated.

olofk, 2019-02-12:
I think it's pretty reasonable to expect that the command you used should work... unfortunately, it's a bit more complicated than this. `--spi_flash_file` is only used to load data into the SPI Flash model during simulations. It doesn't have any effect when running on the real board.

What you should do instead is to create an combined binary file that consists of the FPGA and the software you want to run (e.g. Linux) and put this into the Flash chip. I have prepared a script to help out with this process https://github.com/openrisc/orpsoc-cores/blob/master/systems/de0_nano/utils/flash_helper.sh

Hope this helps and let me know if there are any issues. It's been some time since I used it myself

olofk, 2019-02-12:
Oh, and it looks like `fusesoc pgm` is broken. I need to fix or remove that

imphil, 2019-03-15:
Fixing fusesoc pgm is on my todo list with already some drafts of TCL script laying around. Ping me if you don't see updates on a PR within the next two weeks.

olofk, 2020-12-17:
The pgm command is now marked as deprecated. As for how to run linux on the de0 nano board, I think there should be better instructions at https://github.com/olofk/de0_nano. If this is still of interest, please file a bug report there. Unfortunately I don't think I will have time to look at it anytime soon.

So with that, I will close this issue since I don't think there's anything more we can do on the fusesoc side

closed 2020-12-17 by olofk