olofk/fusesoc #240: Add command line option to list files for a core (and generate if required)

benreynwar, 2019-02-12:
For example:
fusesoc gen run [--export-files] my_core_name output_directory 

This would create the output_directory, and run any generators that need to run placing the
final generation products in the output_directory.
If the export-files flag is set it would also copy all the other files into the output_directory.

It would also write a YAML file to the output_directory with a list of the files (in correct dependency order) and other information that back-end tools might need.

I've been achieving this so far just by importing fusesoc and using the python API, but I want to move to using the command line so I don't have to worry about the GPL.

benreynwar, 2019-02-13:
Just put a pull request #241 that implements this as:

fusesoc run --tool none my_core_name

olofk, 2019-02-14:
Ah ok. I replied to the PR before seeing this. This makes me even more convinced that it should instead be a proper export flag/command that outputs the EDAM file

olofk, 2019-04-15:
Revisiting this now, is `fusesoc run --setup my_core_name` good enough for your use case?

benreynwar, 2019-04-15:
Yes.  I should have closed this pull request.

closed 2019-04-15 by benreynwar