olofk/fusesoc #247: YAML and CAPI2

blacktigersoftware, 2019-03-10:
Not a BUG, but an ISSUE.  I can not for the life of me, figure out how to parse your YAML CAP2 in .NET I get key violations on the 'file' tag.  I tried to represent the file as an object model so as to be serializable, but failed due to variant keys.  What is it that YAML gives that say, something like JSON doesn't give, aside from JSON is supported by just about everybody and I never even had to deal with YAML in any other tool.

For what it's worth, I'm trying to tie FuseSoC with autofpga by reading FuseSoC yaml files and building autofpga build files with a little GUI interface.  I'm doing this in .NET.

I would just be happy if I could figure how to consume these YAML files.  Anybody got any ideas?

olofk, 2019-03-10:
Hi,

I haven't done any .net for many years and never dealt with the yaml/json parser so I'm not sure, but there's one thing I can think of that makes `file` a bit special. It can be _either_ a string (e.g. `path/to/file.v`) or a dictionary/map (e.g. `path/to/file.v : {is_include_file : true, file_type : verilogSource}`). That could be the issue you see

Cool project. Happy to help out where I can. Btw, is that autofpga from @zipcpu or something else?

blacktigersoftware, 2019-03-11:
Yes.  Sorry for sounding a bit 'off'.  Ummm, there isn't a yaml parser in .net, not 'vendor supported' at least.  

But the issue remains.  I went to yaml.org, where there are example implementations.  I tried YamlDotNet, which had the problem.   I guess, I'm more wondering how hard/fast the decision to stick with yaml is?  If that is the case, either I need help finding a parser that work, or I'll have to drop to c++, write a config-file converter to some other format.  Not the best use of anybody's time.

The beauty is that I SHOULD be able to use fusesoc to get IP, and when an instance of that IP is used, create an autofpga spec file.  This generates the needed wrapper that ties all the IP together.  Then, theoretically, I could go back to fusesoc and do the build.   This would totally resemble Vivado Block Diagram technique. 

I currently have an integrated git, an integrated Verilog parser (ANTLR based) which can pop-up customization parameters and a scintilla-based editor.  I want to add build-tasks, mostly spawning processes, to stuff like fusesoc, Icarus, autofpga.  gtkwave.   Maybe even a TCL interpreter.  I had to dump my diagramming, as it sucks and doesn't work that way I want, so I'm looking for that, too.

imphil, 2020-12-18:
I don't think there's much we can do from the FuseSoC side here, apart from switching to another format for CAPI2 files (which won't happen). Closing for now.

closed 2020-12-18 by imphil