olofk/fusesoc #252: Conditionally set paramtype depending on the tool used

stffrdhrn, 2019-03-17:
For a particular use case see mor1kx-generic:
https://github.com/stffrdhrn/mor1kx-generic/blob/master/mor1kx-generic.core#L61

For verilator there is a build stage which needs parameters like `--pipeline=CAPPUCCINO`  or `--enable_mmu` used to control generate blocks in the verilog,  but pulsargs like `--elf-load`, or `--vcd` which should be passed to the verilator simulator binary.

As suggested by @olofk we could use conditional `paramtype` to support this.

If it worked, you should be able to set `paramtype : "tool_verilator? (cmdlinearg) !tool_verilator ( plusarg)"` when defining those parameters

olofk, 2019-03-17:
Yes, this is a bug that I just haven't had time to look into. More generally, I think the problem is that evaluation of conditionals in strings don't work very well. There are other places where this would be useful too, so fixing the general case is preferred

olofk, 2019-04-12:
This is pretty much done now, but there is one problem remaining. Internally, all `-` in argument names are turned into `_`. This means that the verilator model will receive `--elf_load` when using the managed parser. Not really what was intended. Will look into this further