olofk/fusesoc #257: Reduce boilerplate in CAPI2 files

imphil, 2019-05-06:
Currently we have a lot of small(ish) core files for our modules, all of which contain

```
targets:
  default:
    filesets:
      - files_rtl
```

Can we reduce the boilerplate a bit by using the following algorithm?

If no targets section or no targets.default section exists, one is automatically created with all filesets defined in this core file added to it.

olofk, 2019-05-08:
Yes! I've been thinking the same thing. Happy to accept a patch for that.

Also have another related idea about allowing to specify filesets, parameters and similar inline in a target if they are only used by a single target. E.g.

    targets:
      test:
        filesets: [rtl, {files: [tb.v : {file_type : verilogSource}]}]
        parameters : [stop_on_error : {datatype : bool, paramtype : plusarg}]

If something like that would be interesting, we can add a new issue for it

olofk, 2019-09-10:
ok, it should be this simple to fix it

t should be this simple, but apparently not and I have no clue why

    diff --git a/fusesoc/capi2/core.py b/fusesoc/capi2/core.py
    index 0b6db85..09190b4 100644
    --- a/fusesoc/capi2/core.py
    +++ b/fusesoc/capi2/core.py
    @@ -145,6 +145,10 @@ class Core:
                         if not f.logical_name:
                             f.logical_name = str(fs.logical_name)
     
    +        if self.filesets and not self.targets:
    +            self.targets['default'] = Target({'filesets' : list(self.filesets.keys())})
    +            setattr(self.targets['default'], 'name', 'default')
    +
             if self.provider:
                 self.files_root = os.path.join(cache_root,
                                                self.sanitized_name)

What happens is that another core without a target, not in the dependency tree, reports that it's missing a fileset that it doesn't have. Drop the [three attached cores](https://github.com/olofk/fusesoc/files/3597638/cores.tar.gz) into a library, run `fusesoc run --target=tgt t2` and you will see what I mean