olofk/fusesoc #258: vivado: test bench files are added to the top level design sources

hmaarrfk, 2019-05-06:
I'm following the tutorial
https://github.com/olofk/fusesoc/blob/master/doc/tutorials/2-creating_a_core.md

and changed the final line
```
fusesoc run --target=sim --tool=icarus blinky
```
to 
```
usesoc run --target=sim --tool=vivado blinky
```

Unfortunately, it seems that `pps_tb.v` is added to the top level design source and includes non-synthesizable directives (such as `$time`).

I think it should be added to "simulation only" sources.

Doing so manually I get the following commands on the tcl prompt.

```
set_property SOURCE_SET sources_1 [get_filesets sim_1]
add_files -fileset sim_1 -norecurse /home/mark2/cores/gpsemulator/pps_tb.v
```

an issue in fusesoc or edalize.

Seems like there are a few other issues with vivado, in terms of how they like to organize their stuff, but this was the first hiccup I hit.

hmaarrfk, 2019-05-06:
I think this is just my misunderstanding of how to configure things for use with vivado.

For others coming by here: To simulate things, the vivado simulator is called `xsim`, and with that set, the tool is correctly configured.

olofk, 2019-09-10:
Can we close this?

hmaarrfk, 2019-09-11:
I guess so

closed 2019-09-11 by hmaarrfk