olofk/fusesoc #261: Autodetect "system" (VLNV name) when there's only one in the current directory

imphil, 2019-05-06:
End goal:
- Call `fusesoc build` without any system name
- Fusesoc searches for *.core files in the current directory
- If there is only one *.core file, the name of the core is used as system name
- Otherwise an error message is displayed.

Why?
This makes it simpler to call fusesoc in many cases as users don't need to know/remember/look up the name of the system they want to build if they're already in the right directory.

olofk, 2019-09-10:
I just realized one complication. The system name acts as a separator between arguments for the run command and target-specific parameters (plusargs, defines, etc). There are probably ways to get around that but until a solution for that pops up I don't think we should pursue this.

The only thing that comes to mind is implementing the `core.target` solution in https://github.com/olofk/fusesoc/issues/260 and require that users at least specify a target. e.g. `fusesoc run --run_parameter .target --target_parameter`. Not sure that would improve the situation all that much though

olofk, 2025-02-21:
I'm closing this as wontfix, as I think it's just not worth the added complexity to avoid having to write the VLNV

closed 2025-02-21 by olofk