olofk/fusesoc #263: Only set a vlogdefine, do not assign a value to it

imphil, 2019-05-07:
Goal: have fusesoc pass a define as "being set/defined" to a tool, do not assign a value to it.

Currently we have code like this:
```yaml
parameters:
  SYNTHESIS:
    datatype: str
    paramtype: vlogdefine

target:
  lint:
    parameters:
      - SYNTHESIS=1
```

Switching `SYNTHESIS=1` to `SYNTHESIS` will only make the vlogdefine available to the tool, but not set it. This is documented and intended behavior and generally makes sense.

For the "only define, don't assign a value" use case we need to extend the syntax a bit.

In the `target` section, I don't see a way to distinguish between "make vlogdefine available", "set vlogdefine without value", and "set vlogdefine to a given value".

I'd instead propose a new datatype `define` or `isset` or something like that (no good ideas come to mind right now). Assigning a truthy value (like 1 or true) to it would result in the define being set and passed to the EDA tool. Assigning a falsy value to it will not pass it to the EDA tool.

olofk, 2019-05-10:
IIRC I experimented with this a while ago to have vlogdefine with datatype bool only define the var without assigning a value. The problem as I remember it was that some tools wouldn't let me set a define without a value. Might be possible to work around that, but I stopped investigating as I couldn't really find a valid use case for it. Do you actually need to have the define set without a value, or is it mainly for symmetry?

olofk, 2020-02-29:
Is this still valid or can we close?