olofk/fusesoc #264: Warn or error if fileset includes files above core file

imphil, 2019-05-08:
We should warn or error out if users have a core file which specifies a file path in a fileset with a relative path above the core file, e.g.

```yaml
filesets:
   file_rtl:
     files:
      - ../other_module/something.sv
```

Using fusesoc like this can work (if other_module for some reason part of the build as well, e.g. because of a dependency), but it doesn't work in the general case.

We should warn loudly or error out if this behavior is detected.

Sounds good @olofk? I'll open a PR then.

olofk, 2019-07-02:
Would it be enough to just check for `..` as the two first characters? Seems like it but I'm afraid I've missed something

imphil, 2019-07-02:
Checking for ".." as first file name component is sufficient (unless people use absolute paths. Maybe that should be disallowed as well)

imphil, 2020-12-18:
I opened https://github.com/olofk/fusesoc/pull/459 to issue a deprecation warning when paths outside of the core directory are found, with the intention to add the error in the next major release.

olofk, 2021-01-26:
We can close this now. The deprecation notice should be enough of a reminder for us. Reopen if you disagree

closed 2021-01-26 by olofk