olofk/fusesoc #272: Add support for Xrun (Cadence)

Alex-Mann, 2019-05-10:
Noting this here to start a trail for including support for Xrun, the simulation command line utility by Cadence.

I am hoping to tackle some of this as I have some free time, and this thread can serve for others to keep track of the status and/or contribute.

olofk, 2019-05-11:
This would be a very welcome addition. I don't have any experience with the tool but let me know if you need general info about writing backends

imphil, 2019-05-21:
xrun is essentially irun, so the work in https://github.com/olofk/edalize/pull/29 will be the first step. From what I've seen so far the command line arguments are mostly the same, so we should try to support both tools with either the same backend or at least shared code. (The renaming wasn't a terribly great idea by Cadence ...)

Alex-Mann, 2019-05-21:
I agree. If we can support both tools with shared code, that is the ideal situation. From my usage so far, both have a lot of overlap, so I think this is possible. Thanks for pointing to the above issue, I've skimmed lightly and will go deeper later today. I can provide some insight into options and such later on so that we can get the API right.

Alex-Mann, 2019-05-21:
I agree. If we can support both tools with shared code, that is the ideal situation. From my usage so far, both have a lot of overlap, so I think this is possible. Thanks for pointing to the above issue, I've skimmed lightly and will go deeper later today.

olofk, 2019-05-22:
Sharing code is good, but since backends are quite small it might make sense to keep "mostly the same" backends separate. An example of this is modelsim and riviera pro which use the same command names but have slightly different behaviour when it comes to how simulations are stopped and how VPI modules are compiled. OTOH, we will soon have a bunch of backends using yosys or synplify which is something we should look in to sharing between backends

cmarqu, 2019-05-24:
The Incisive to Xcelium migration rules are: everything that has an ``nc`` (case-insensitive) in an option or environment variable name is replaced with ``xm`` for Xcelium. ``INCA_libs`` is now ``xcelium.d``, ``irun`` is now ``xrun``.

olofk, 2020-05-04:
If I'm not mistaken, we should be able to close this now thanks to https://github.com/olofk/edalize/commit/6093ee9a40c74673f0fc2f8c0d464141311f60cd Please reopen if I'm mistaken

closed 2020-05-04 by olofk