olofk/fusesoc #274: Inconsistent parameters: align sim closer to run

imphil, 2019-05-21:
Can we align the `fusesoc sim` arguments a bit closer to their `run` equivalents?

```
usage: fusesoc sim [-h] [--no-export] [--sim SIM] [--setup] [--build-only]
                   [--force] [--keep] [--target TARGET]
                   [--testbench TESTBENCH]
                   system ...

positional arguments:
  system                Select a system to simulate
  backendargs

optional arguments:
  -h, --help            show this help message and exit
  --no-export           Reference source files from their current location
                        instead of exporting to a build tree
  --sim SIM             Override the simulator settings from the system file
  --setup               Only create the project files without running the EDA
                        tool
  --build-only          Build the simulation binary without running the
                        simulator
  --force               Force rebuilding simulation model when directory
                        exists
  --keep                Prevent rebuilding simulation model if it exists
  --target TARGET       Override default target
  --testbench TESTBENCH
                        Override default testbench
```

```
$ fusesoc run --help
usage: fusesoc run [-h] [--no-export] [--build-root BUILD_ROOT] [--setup]
                   [--build] [--run] [--target TARGET] [--tool TOOL]
                   system ...

positional arguments:
  system                Select a system to operate on
  backendargs           arguments to be sent to backend

optional arguments:
  -h, --help            show this help message and exit
  --no-export           Reference source files from their current location
                        instead of exporting to a build tree
  --build-root BUILD_ROOT
                        Output directory for build. Defaults to build/$VLNV
  --setup               Execute setup stage
  --build               Execute build stage
  --run                 Execute run stage
  --target TARGET       Override default target
  --tool TOOL           Override default tool for target
```

- It's `--build` in `run`, and `--build-only` in `sim`. I'd rename `--build-only` to `--build` (and deprecate the other one by issuing a warning, but leaving it supported for now.)
- Is there an equivalent for `--keep` and `--force` in the `run` subcommand (or should there be?)

olofk, 2019-05-22:
the build and sim commands are really supposed to be deprecated already so I don't want to touch them but just keep them as long as it's feasible to not break people's old scripts. For the build command it can be argued that the equivalent `run --setup --build` isn't all that ergonomic. The problem here is the inconsistency between the FPGA backends whether to program the board or not during the run phase.

If the ergonomic aspect of `run --target=<target> <core>` is the issue at hand I'm much more include to add a short-hand notation such as `run <core>.<target>`

I'm not sure `--force` and `--keep` makes any sense together. I always saw them as opposites and have honestly no idea (without looking at the code base) what happens if you specify both at the same time. Oh.. or do you mean equivalents for each one? In that case, I'm not sure. `--keep` can be performed by `--build` (and optionally `--run`). `--force` is the default behaviour when the setup stage is run

olofk, 2020-05-04:
Can we close this now? And eventually remove the sim command

imphil, 2020-07-27:
Let's close this. build and sim have been deprecated for quite a while, and I've opened #429 to get them removed.

closed 2020-07-27 by imphil