olofk/fusesoc #275: Share tool_options between core files

imphil, 2019-05-22:
It would be nice to have a way to share tool_options between core files, e.g. in a configuration file.

For example, VCS users will often set the `-licqueue` parameter for all their simulations.

I'll follow up with a more concrete design proposal later, opening this issue now to get it tracked at least.

wallento, 2019-05-22:
I am wondering, doesn't that more relate to the tool and not the core? So that would mean its more like `-f` in some tool that reads options from a file or so?

imphil, 2019-05-22:
@wallento Yes exactly. `tool_options` is currently a mixed bag of 
- options needed to configure the tool in general, and 
- options to configure the tool for this specific simulation/tool run. 

These options end up in the same command line argument ultimately, but we should look for a way to specify them in different places.

olofk, 2019-05-22:
An improved configuration file format would indeed be nice where default options of different kinds can be set. Options specific to the tool or environment are especially suited for that. Happy to discuss and take ideas for design of that

olofk, 2019-05-29:
btw, a somewhat hacky solution that I think could work in the meantime would be to add a new core with

`targets : {default : {tool_options : {vcs_options : [-licqueue]}}}` (syntax not tested) and depend on it from your toplevel or some common core

olofk, 2020-12-17:
@imphil can we close this one now, with the things we learned from https://github.com/lowRISC/opentitan/pull/4491?

imphil, 2020-12-17:
Yeah, let's call it a day for this one.

closed 2020-12-17 by imphil