olofk/fusesoc #279: Run synthesis only.

m-kru, 2019-06-18:
Is there any way to make fusesoc stop after synthesis, without P&R? In `fusesoc --help` I can see following options:
```
    build               Build an FPGA load module
    init                Initialize the FuseSoC core libraries
    pgm                 Program an FPGA with a system configuration
    fetch               Fetch a remote core and its dependencies to local
                        cache
    core                Subcommands for dealing with cores
    list-cores          List available cores
    core-info           Display details about a core
    gen                 Run or show information about generators
    list-paths          Display the search order for core root paths
    library             Subcommands for dealing with library management
    run                 Start a tool flow
    sim                 Setup and run a simulation
    update              Update the FuseSoC core libraries
```
It looks like it is not possible, but maybe there is an option to achieve it via some config?

olofk, 2019-06-19:
Hi,

Yes. This is planned. Just haven't gotten working on it yet. As a first step this will be implemented as backend options in the synthesis/P&R backends. Currently, the icestorm backend has an option to set `pnr` to `arachne` or `next`. The idea is to add a `pnr` backend option and allow to set the vallue to `none` to have it stop after synthesis

olofk, 2019-06-19:
Is there a backend in particular (like vivado, quartus, icestorm...) you're interested in? In that case I could start with that one

olofk, 2019-06-19:
I made a (somewhat hacky) addition to the vivado backend now. Get the latest edalize and you can now run with `--pnr=none` on the command-line or set the tool option in your .core file

`fusesoc run --target=yourtarget yourcore --help` should list `--pnr` as an option if it uses the vivado backend

m-kru, 2019-06-29:
Ok, it works as you have described.  I think it can be closed if you are happy with the implementation. From user perspective it does what it should.

olofk, 2019-06-29:
Cool. I think it's good enough. Improvements can go in later

closed 2019-06-29 by olofk