olofk/fusesoc #280: Bool parameter for boolean VHDL generic produces wrong tcl command for Vivado.

m-kru, 2019-06-18:
You can check it here: [fusesoc_example](https://github.com/m-kru/fusesoc_example). Just run `fusesoc --cores-root . run --target custom_target_2 ::fusesoc_example` and check file `build/fusesoc_example_0/custom_target_2-vivado/fusesoc_example_0.tcl` line 9.

There is:
`set_property generic {TOP_ADD_DELAY=1 } [get_filesets sources_1]`
Should be:
`set_property generic {TOP_ADD_DELAY=true } [get_filesets sources_1]`

I could fix this if you tell me how/where the fix should be placed? In fusesoc or edalize?

olofk, 2019-06-19:
Yes, you're right, this is a bug. Thanks for reporting. It's in edalize, but I wasn't sure where to actually fix it so I had to look it up, and once I found it I made a fix in https://github.com/olofk/edalize/commit/57fb0d465c72bed963f4cc6a1083395a8418951e . Could you try the latest edalize and close the bug if it works for you?

m-kru, 2019-06-20:
I will let you know on Monday as I don't have any access to my machine with EDA tools.

m-kru, 2019-06-24:
I have tried. I have installed latest master commit from edalize with pip and I get following error:
jinja2.exceptions.TemplateNotFound: `vivado/vivado-synth.tcl.j2`
In `/usr/lib/python3.7/site-packages/edalize/templates/vivado` indeed there is no `vivado-synth.tcl.j2` file. Looks like setup.py also needs to be updated?

olofk, 2019-06-25:
Doh! You're right. I added it now. Another attempt?

m-kru, 2019-06-25:
There is one more issue. Right now edalize generates:
`set_property generic {TOP_ADD_DELAY=True } [get_filesets sources_1]`
should be:
`set_property generic {TOP_ADD_DELAY=true } [get_filesets sources_1]`
Note the lowercase `t` letter for `true`.

olofk, 2019-06-25:
Does this matter though? VHDL is case insensitive and none of the other tools seems to have issues with the capital T. I'll change if it doesn't work of course. Just curious

m-kru, 2019-06-26:
I test with Vivado 2018.3, with `True` the build fails, with `true` it is successful. I am also astonished. Vivado output doesn't explicitly inform that the problem is with `True` as the out in case of failed build is as follows:
```
# open_run impl_1
ERROR: [Common 17-69] Command failed: Run 'impl_1' has not been launched. Unable to open
INFO: [Common 17-206] Exiting Vivado at Wed Jun 26 07:14:52 2019...
make: *** [Makefile:6: fusesoc_example_0.bit] Error 1
```
To make the build successful I need to replace `True` with `true`.
This is Xilinx.
[image]

olofk, 2019-06-26:
*Sigh* Oh well, this is why edalize exists after all, to save everyone from having to deal with all the peculiarities of each EDA tool.

Hope this is fixed now

m-kru, 2019-06-26:
Works.

closed 2019-06-26 by m-kru