olofk/fusesoc #282: How to depend on different descriptions of the same module in different projects?

m-kru, 2019-06-30:
Lets imagine that one would like to create library with modules described in both VHDL and Verilog. Is it possible to use single `.core` file for such module? Here is a tree of what I would like to achieve:
```
.
├── LICENSE
├── mux_2_to_1
│   ├── mux_2_to_1.core
│   └── rtl
│       ├── verilog
│       │   └── mux_2_to_1.v
│       └── vhdl
│           └── mux_2_to_1.vhd
└── README.md
```
I saw that you sometimes do in the same way, for example here [i2c](https://github.com/olofk/i2c). However in [fosesoc-cores](https://github.com/fusesoc/fusesoc-cores) there is only fileset for Verilog in `.core` file.

In upper modules I would like to depend on this module and be able to choose if I want to use VHDL or Verilog filesets. It is for sure possible with 2 separate `.core` files and different names for the module, but I think it is cluttered way.

If it is possible, is there any example that would show how to depend on such module and choose its description?

olofk, 2019-07-02:
This is currently not possible with FuseSoC but I offer two answers that might help.

1. This type of functionality is planned for the next release of FuseSoC and will be tracked here https://github.com/olofk/fusesoc/issues/268 The idea is to allow users to specify flags on the command line to enable and disable functionality. Very much like the implicit tool_* flags and is_toplevel which already exists.

2. Depending on your actual use case for this, there might be options. If you just need one of the implementations for a specific tool, you can define two filesets and in the target add them as e.g.
`filesets : ["tool_ghdl? (vhdl_files)", "!tool_ghdl? (verilog_files)"]` It might not help in your use case, but still worth mentioning.

m-kru, 2019-07-03:
I see one solution, this looks clear to me and solves my problem. What about possibility to specify not only module with `depend` but also filesets? Right now when one module depends on another one it uses filesets from `default` target for that module. This could be left as default behavior. For example:
`    depend : ["vendor:lib:module_x"]`
would fetch filesets for `default` target of module_x, however
`    depend : ["vendor:lib:module_x-default_vhdl"]`
would fetch filesets for `default_vhdl` target. `-` as a separating character can be replaced with something else like `*`, `/` etc.
What do you think? 

Flags on the command line are good for enabling/disabling functionalities globally. However in my case users module may depend on few modules and there might be a need to choose different targets of these dependencies, not only `default`.

olofk, 2019-09-10:
I definitely see the appeal with being able to depend on a specific target but I want to make sure we have thought of everything before going down that route. The more I think about, the more I'm starting to like it, so I'm not fundamentally opposed to it.

In the meantime, I want to mention that there is now preliminary support for useflags in the useflags branch. Would love to know what kind of situations this improves or not

olofk, 2020-01-08:
I realized that there is a problem with depending on custom targets. There can only be one instance of each core in the dependency tree and if we allow custom targets we can end up in situations where one core depends on target1 in a core and another one depends on target2 in the same core. I'm afraid we can't solve that with the current dependency manager.

So in the short term I think we will have to live with the global use flags which is now in the latest release, and in the long term I really really really want a new dependency manager that can deal with this in a better way. Either one written from scratch or possibly use something like pkgcore

imphil, 2020-12-18:
Discussion around "inheriting" from targets other than default is in #414, let's continue it there.

closed 2020-12-18 by imphil