olofk/fusesoc #284: What about adding filesets_groups?

m-kru, 2019-07-05:
Issue:
A lot of targets that share the same HDL sources but have different pinouts, synthesis and implementation strategies etc. 

Right now I repeat most of the filesets part for each target, like this:
```
 targets:
    target_0:
        default_tool: vivado
        tools:
            vivado:
                part: xc7a200tfbg484-2
        toplevel: my_top
        filesets:
            - target_0_pinout
            - target_0_settings
            - fileset_a
            - fileset_b
            .
            .
            .
            - fileset_z
  target_1:
        default_tool: vivado
        tools:
            vivado:
                part: xc7a200tfbg484-2
        toplevel: my_top
        filesets:
            - target_1_pinout
            - target_1_settings
            - fileset_a
            - fileset_b
            .
            .
            .
            - fileset_z

  target_N:
        default_tool: vivado
        tools:
            vivado:
                part: xc7a200tfbg484-2
        toplevel: my_top
        filesets:
            - target_N_pinout
            - target_N_settings
            - fileset_a
            - fileset_b
            .
            .
            .
            - fileset_z 
```
It leads to lots of repetitions and long .core files.
Imagine something like this:
```
filesets_groups:
  my_group:
     - fileset_a
     - fileset_b
     .
     .
     .
     - fileset_z    

 targets:
    target_0:
        default_tool: vivado
        tools:
            vivado:
                part: xc7a200tfbg484-2
        toplevel: my_top
        filesets:
            - target_0_pinout
            - target_0_settings
        filesets_groups:
            - my_group
  target_1:
        default_tool: vivado
        tools:
            vivado:
                part: xc7a200tfbg484-2
        toplevel: my_top
        filesets:
            - target_1_pinout
            - target_1_settings
        filesets_groups:
            - my_group

  target_N:
        default_tool: vivado
        tools:
            vivado:
                part: xc7a200tfbg484-2
        toplevel: my_top
        filesets:
            - target_N_pinout
            - target_N_settings
        filesets_groups:
            - my_group
```

Such construct is shorter and easier to manage. What do you think?
If any mechanism similar to this already exists let me know, and forgive me but I was not able to find it in any example or documentation.

olofk, 2019-07-18:
You can use yaml anchors to avoid repeating the default_tool, tools and toplevel parts. But if you just want to change part of a list, that won't work. I had the same issue [here](https://github.com/tinyfpga/TinyFPGA-Bootloader/blob/master/TinyFPGA-Bootloader.core) and I agree it's very annoying for these cases.

Your solution makes sense, but I would like to think it through to see that it doesn't clash with anything else. Perhaps this should be generalized as I could see the same thing with e.g. parameters or generators.

One, slightly related, thing I have thought about is to allow anonymous inline filesets (and parameters, generators, etc) in the targets for things that only appear once in the core file. Imagine something like this

    targets:
      target_N:
        filesets:
          - fileset_a
          - fileset_b
          - fileset_c
          - {files : [pinout_N.xdc : {file_type : xdc}, settings_N.tcl : {file_type : tclSource}]}

It won't solve your original problem, but will save some extra typing

m-kru, 2019-07-19:
Anonymous inline filesets would also be nice improvement. Let me know if you come up with something for grouping functionality!

olofk, 2019-07-20:
I'm leaning towards features to append (and possibly remove) lists. This is something I have looked at before but postponed due to other features.

The idea is to have separate keys in the targets that tells FuseSoC which extra filesets to include (or again, possibly remove) from the ordinary filesets list. Will work just the same for other lists, such as parameters or generators.

Using the bitbake syntax which is one candidate to reuse it would look like this

    targets:
      target_n:
        filesets: [fileset_a, fileset_b, fileset_c]
        filesets_append: [target_N_pinout, target_N_settings]

This will not help much by itself, but coupled with yaml anchors we could now write

    targets:
      dummy_target_or_default_whatever_makes_sense: &base
        default_tool : vivado
        filesets: [fileset_a, fileset_b, fileset_c]
        tools:
          vivado:
            part: xc7a200tfbg484-2
        toplevel: my_top

      target_1:
        <<: *base
        filesets_append: [target_1_pinout, target_1_settings]

      target_n:
        <<: *base
        filesets_append: [target_N_pinout, target_N_settings]

 I think the bitbake syntax would work fine but googling a bit I have seen other syntax example like `filesets+`. Looks a bit nicer in my opionion, but I think the corresponding `-` would conflict with yaml syntax, so using letters might be a good idea.

I believe the append solution would cover the most common use case, namely when you just want to add an additional fileset. Typical example is having the `default` target with the core RTL and then some board-specific targets which adds wrappers and a simulation target that adds a testbench.

Thoughts?

m-kru, 2019-07-22:
It looks like clean and compact solution, and solves retyping problem so I am in favour.

olofk, 2019-07-23:
I made a quick PoC with just support for appending filesets [here](https://github.com/olofk/fusesoc/tree/append). It seems to work, but I would appreciate if you could do a quick check to see if it works as you intended before I push this forward for other types of lists

m-kru, 2019-07-23:
Ok, I have checked and it does what I have expected it to do.

Right now I wonder about one thing. Is there any scenario where user would need to prepend filesets to list instead of appending?

olofk, 2019-07-23:
Cool. I'll clean it up and submit a proper patch when I get the time.

As for prepending, maybe, but let's add that when there's a valid use case for it. I'm very reluctant to adding things that doesn't have a user

m-kru, 2019-07-24:
I totally agree. The less code, the better.

olofk, 2019-07-26:
Alright. I just pushed a commit which makes all lists appendable. Please let me know if it works and feel free to close or complain otherwise

m-kru, 2019-07-27:
Works, I am closing. By the way, I do not want to be rude, but when will you find time to answer issues and pull requests? Both from [FuseSoc](https://github.com/olofk/fusesoc) and [edalize](https://github.com/olofk/edalize).

closed 2019-07-27 by m-kru