olofk/fusesoc #285: Inconvenience encountered - FuseSoc + GHDL + UVVM

m-kru, 2019-07-05:
I use [GHDL](https://github.com/ghdl/ghdl) as simulation backend and [UVVM](https://github.com/UVVM/UVVM) as test bench framework. I wanted to update this really old [uvvm_util-1.2.0.core](https://github.com/openrisc/orpsoc-cores/tree/master/cores/uvvm) `.core` file for [UVVM](https://github.com/UVVM/UVVM) and move it to the [fusesoc-cores](https://github.com/fusesoc/fusesoc-cores).
However, I have realized, that when I run test benches with FuseSoc and GHDL backend, [edalize](https://github.com/olofk/edalize) ghdl script makes GHDL always compile all `.vhd` files needed to run a test bench. Compiling UVVM is not that short, and if  you want to run lots of short test benches, recompiling UVVM `.vhd` files takes 90% of the time you wait for the results.

My current solution:
I have compiled UVVM as library for GHDL and in `.core` files I add option for GHDL tool:
`analyze_options: [-P/path/to/ghdl/libraries]`
It is easy to notice that this solution has one drawback. Hard coded path for libraries, what makes `.core` files unportable. 

I have come up with one idea, but I am not sure if it is the best one.
A patch to the GHDL could be added. With that patch there would be an option to ask GHDL what libraries are already compiled and seen by GHDL. However here the new question arises: How should edalize backend know that this particular dependency is a library that can be already compiled for chosen target tool? I also have one idea here. Add this information to the `.core` file for this particular module. Example:
```
CAPI=2:

name: btvis:uvvm:uvvm_foo

potential_library:
    tools:
        - ghdl

filesets:
 ...
 ...
 ...
```

What do you think? Do have any ideas how this could be solved?

m-kru, 2019-07-29:
@olofk ping

olofk, 2019-08-26:
One thing that might work is to use a generator for this. On the first run it would compile and cache the library in `~/.cache/fusesoc/generated/` and on subsequent runs it would just pick this library up. There are a few corner cases to be aware of here

1. FuseSoC will hardcode the directory name in `~/.cache/fusesoc/generated/` to `vendor_library_name-generatorname_version`, so if you have a different toplevel it will not be able to reuse the cache
2. When do you invalidate the cache? Easiest is to just remove the dir manually. Another option could be to pass an environment variable that the generator picks up and decides if it should regenerate. A third option, that requires use flags would be to add a key to the generator section to force a regeneration and set that key as a use flag on the command-line

olofk, 2019-08-26:
Oh, and before generators I used a pre_build_script to do something similar when compiling Xilinx simulation libraries. That method is harder to make portable though

m-kru, 2019-08-26:
I have walked around the problem of recompiling test bench frameworks in this tool: [fsva](https://github.com/m-kru/fsva)

olofk, 2019-08-26:
Cool. Then we perhaps should close this?

m-kru, 2019-08-26:
Well, we can. However, it looks like FuseSoc can generally be improved by recompiling only these HDL files that have changed, but you are probably aware if this :).

closed 2019-08-26 by m-kru