olofk/fusesoc #286: Problems compiling FuseSoc

lorenzdbs, 2019-07-20:
Hi, 

I'm following [this](https://github.com/kdgwill/OR1K/wiki/OpenRisc-tools-Install) tutorial in order to install the OpenRisc or1k tool-chain but I've ran into an issue when I've had to compile and install FuseSoc.
Once I've to run the `autoreconf -i` command it tells me that _configure.ac_ or _configure.in_ are required. How can I generate it? As I saw that the other package as OpenOCD or Opensim are provided with this file I thought that it is missing from the git repository where I cloned it from: could it be?
Thanks to all those who can help me,

Lorenzo

olofk, 2019-07-20:
Hi,

You managed to follow an ancient tutorial. FuseSoC hasn't had autotools support for many years. The easiest way to install FuseSoC nowadays to run `pip install fusesoc`

lorenzdbs, 2019-07-21:
Thank you so much @olofk for the help. I've read it in the _README.rst_ but I was anyway waiting for the answer of someone more expert.
Thank you again, have a nice day

closed 2019-07-21 by lorenzdbs