olofk/fusesoc #287: Isn't sim command redundant?

m-kru, 2019-07-22:
The purpose of sim command is to `sim                 Setup and run a simulation`.  However, I feel that this is a bit redundant. Simulation can also be run with `run` command by specifying proper target and tool. From main.py it looks like the target for sim  is fixed to [sim](https://github.com/olofk/fusesoc/blob/670c618d6f0e36e3e039bb51c86964061f2c9e88/fusesoc/main.py#L369). There is also optional argument   `--keep                Prevent rebuilding simulation model if it exists` that seems not to work, at least with GHDL. It does prevent rebuilding simulation model if it exists, but it fails when model needs to be built. 

What do you think about removing `sim` command from fusesoc?

olofk, 2019-07-23:
That's correct. `sim` is redundant and `run` should be used instead. The background is that the `build` and `sim` commands were designed for `CAPI1` where there was two very distinct pipelines for synthesis and simulation. With CAPI2, I tried to unify this a bit and also introduce possibilities to support more types of tools (e.g. formal and linting tools). There are still users of the two original commands however, and as long as the support burden is small enough, I want to keep them for backwards compatibility. Eventually sim and build will both go away though.

As a first step, I think it would be good to just add a warning to tell people to use the run command instead

m-kru, 2019-07-23:
Ok, I can do that.

olofk, 2019-07-30:
Closing this as #288 is merged

closed 2019-07-30 by olofk