olofk/fusesoc #291: Problem with processing order with generated cores.

m-kru, 2019-07-27:
Maybe I am doing something wrong, but it looks like files from generated cores are processed as the last one, before toplevel. I need generated files in an entity that is not top entity. However, no matter how I write `.core` file, generated vhdl files are always processed as the last one (before toplevel). I have tried different configurations, like splitting filesets, so that the first fileset in the target depends on the generator and the last one fileset includes `.vhd` file that needs the generated files, but without success.

As processing order plays a crucial role in HDL world I stuck.

olofk, 2019-07-27:
As you say, ordering is important and FuseSoC is strict about that. By default, generated cores will come after the core that requested the generation. However, you can set `position : first` to make the generated core appear before regular cores

https://fusesoc.readthedocs.io/en/master/capi2.html#generate

Does that help?

m-kru, 2019-07-27:
It works, but does not solve my issue, because I need to process some other VHDl files as firsts. The best solution for me would be to processes these generated files before files that are in the fileset that depends on the generator.

olofk, 2019-07-28:
Right, so would a `prepend` option solve your issue? I considered adding it originally but was not sure there was an actual use case for it. Happy to do it now if it's needed.

m-kru, 2019-07-29:
`position : prepend`? Yes, that would solve my issue.

m-kru, 2019-08-19:
The problem I have encountered was due to bad dependency description. I am closing as not actual.

closed 2019-08-19 by m-kru