olofk/fusesoc #292: CAPI2 file types

djcopley, 2019-07-29:
I have not been able to find one go-to place for file_type parameters. Is there a place with a list of all possible CAPI2 types. To make things more confusing, I found the following snippet in the edalize vivado backend:

```
Vivado Backend
A core (usually the system core) can add the following files:
- Standard design sources
- Constraints: Supply xdc files with file_type=xdc
- IP: Supply the IP core xci file with file_type=xci and other files (like .prj)
      as file_type=data
```

File type xdc, xci, and data yield `unknown file type '[xdc, xci, data]'`(the brackets indicate a selection of one of the three, not all three).

On a separate note, I am having trouble getting the pnr option working. I have installed the latest version of fusesoc and edalize using pip, to no avail. While it is likely user error, any suggestions are appreciated. My target:

```
targets:
  synth:
    default_tool: vivado
    filesets: [src]
    tools:
      vivado:
        part: xc7a200tfbg676-2
        pnr: none
    toplevel: topblock
```

For context, I am trying to package a xilinx bist using fusesoc.

djcopley, 2019-07-29:
Questions cont'd (sorry)

1. Currently, I am specifying `{file_type : user, copyto ./}` for simulation files like data and waveform config files, then sourcing a tcl script (in xsim options -t init.tcl) that adds the files to the simulation. 

2. Furthermore, the tcl script that edalize generates (run-gui.tcl) adds all the signals to the waveform viewer. This is particularly annoying when you are specifying a wcfg file. I have tried to override this behavior with scripts and hooks (no success).

3. Can I launch the simulator in gui mode from the command line (without `make run-gui`)?

m-kru, 2019-07-29:
I am not FuseSoc maintainer but I have used it a lot recently. `xdc` and `xci` file types work for Vivado as I use them heavily. I don't see in the edalize source code any 'data' file type?

djcopley, 2019-07-29:
@m-kru https://github.com/olofk/edalize/blob/master/edalize/vivado.py#L18

Jbalkind, 2019-07-29:
They do work for me too (though don't let Olof see that I used paths that he doesn't like...):

https://github.com/Jbalkind/openpiton/blob/jbalkind_fusesoc_sims/piton/design/rtl/system.core#L19-L32

djcopley, 2019-07-29:
So something like this should work...
```
filesets:
  src:
    files:
      - src/top.vhd : {file_type : vhdlSource-93}
      - src/test.xdc : { file_type : xdc }
```
and not yield `../src/icr-team.com__fir_filter_1.0/src/test.xdc has unknown file type 'xdc'`?

@Jbalkind, I would have fully expected xdc and xci file types to work (hence my confusion when they didn't).

m-kru, 2019-07-29:
@djcopley What backend do you use? `xdc` works for Vivado but does not work for xsim (Xilinx simulator).

djcopley, 2019-07-29:
@m-kru Oh. My. God. Yep, I was using the fileset in my sim target. Works when I isolate my constraints. Facepalm :/

djcopley, 2019-07-29:
User error. Thanks @m-kru and @Jbalkind.

closed 2019-07-29 by djcopley