olofk/fusesoc #295: Problem with updating library after removing it from .local/share/fusesoc

m-kru, 2019-07-30:
If user removes `my_lib` directory from `.local/share/fusesoc`, but leaves entry for that library in `fusesoc.conf`, and then runs `fusesoc update` , `my_lib` is not fetched (is not downloaded to the `.local/share/fusesoc`).
I am not sure if such behavior is desirable, but I personally find it confusing. I think `fusesoc update` should fetch latest master for all libraries defined in `fusesoc.conf` file

olofk, 2019-08-26:
I think I agree with you. Just trying to think if there are any cases where you don't want to fetch it. Maybe fetch it by default and add a flag that ignores all libraries which aren't already fetched

m-kru, 2019-08-26:
In my humble opinion, the behavior of `fusesoc library update` should be very simple. Namely, update all libraries that are in `fusesoc.conf` file. By update I mean clone the repository, if it is not in the `.local/share/fusesoc` directory or pull the latest master if it is already there. If user does not want some library anymore, then delete it from `fusesoc.conf` (btw. there should probably be some command for this, instead of doing it manually).

nevercast, 2020-06-22:
Just encountered this when I depend on https://github.com/davidthings/tinyfpga_bx_usbserial. My project fusesoc.conf is as follows: https://github.com/nevercast/tinysoc/blob/138f1d1def95bff8e5eee929c5e590eb5a3cf4eb/fusesoc.conf 

I don't wish to commit `fusesoc-libaries/tinyfpga_bx_usbserial` to git, but if I remove it then anyone else will face the problem of being unable to build my core.

I expected `fusesoc libraries update` to at least reclone it to save me a hot minute, just as described by @m-kru. But I also expected the target that depends on the library to also update it. I can probably live without that last one, but it would be a really nice flow to at least check if libraries are not cloned when a target fails to find a dependency.