olofk/fusesoc #296: Functionality for automating simulation based verification.

m-kru, 2019-07-31:
I am currently working on the script that will help me automate simulation based verification. It will basically filter all cores in a given directory and choose only these cores that have targets following one of the string patterns `tb_* `or `*_tb`. It will then run all these targets, save output from test benches to files to some directory and print some summary. I have realized that this feature actually could be a part of FuseSoc. For example `fusesoc testbench lib_name/path_to_lib?` would execute all test benches for given library/project.

Do you want it as a part of FuseSoc or should I keep it as my custom script?

m-kru, 2019-08-20:
After some tries I have realized that it would require more changes than I thought. I have decided that it would be better to implement some wrapper for FuseSoc, so that it doesn't get over bloated. With such approach the main goal of FuseSoc remains the same: efficient single target building/running.

Btw. If you are interested here is the tool: [fsva](https://github.com/m-kru/fsva).

closed 2019-08-20 by m-kru

olofk, 2019-08-26:
Wow! This is great! Exactly what I was hoping someone would do eventually :) I'll try to take a look at it in detail soon. Thanks