olofk/fusesoc #298: --build implies --setup only if it doesn't exist yet

imphil, 2019-08-01:
Consider the following scenario:

1. Make sure you have a clean `build` directory.
2. Run the following command on a Verilator-based design:
   ```
   fusesoc run --target=sim --build your:design
   ``` 
   This will run the setup stage first, and then the build stage.
3. Edit a source file.
4. Now run the same command again:
   ```
   fusesoc run --target=sim --build your:design
   ...
   make[1]: Nothing to be done for 'default'.
   ``` 
   Only the build stage is run, and the makefile detected no changes in its dependent files, so the simulation isn't rebuilt.

Expected result: the simulation is actually rebuilt (I changed a source file).

This problem is present since https://github.com/olofk/fusesoc/commit/ae9cdc3444aaad6eb718283fbc5fe37ad9df9a63.

Possible solutions:
- Workaround: *Always* execute --setup with --build.
- Solution: Track all source file dependencies in the Makefile. This would actually avoid recompiling if no sources changed.

I hope to find time to look at this in more detail some time next week, for now this issue is just to get feedback and raise awareness.

olofk, 2019-08-01:
I always wanted source file dependency tracking but never got around to do it :( As you say, that's the correct way and will also save a lot of rebuilding in other cases (note, I think by the time we do that, we should also look into moving the src dir under the work_root. Pretty sure the only thing that will break are hard coded paths in old scripts).

Now, should we implement the workaround or leave it as it is but document it? I'm trying to remember if there was a case for not always doing --setup, but I can't think of anything

olofk, 2020-01-08:
Should we implement the workaround for now?

imphil, 2020-01-08:
Yes, that would be good.

olofk, 2020-01-08:
ok, done. Close if it works

olofk, 2020-01-31:
@imphil Have you had a chance to test this? I believe it should work but would be great to have it confirmed

closed 2020-12-18 by imphil