olofk/fusesoc #300: Fetching core and its dependencies.

m-kru, 2019-08-23:
`fusesoc fetch core` can fetch files for remote core. However, if this core has other remote cores in its dependencies, it would be nice to have an option to automatically fetch them. 
For example, add `--with_dependencies` flag to fetch command:
`fusesoc fetch --with_dependencies core`.

olofk, 2019-08-26:
That's a good idea. I thought it always fetched dependencies (since the description says Fetch a remote core and its dependencies to local cache)

One problem with this however is conditional dependencies, like if a dependency only appears in one target, or if two different tools have different dependencies. The dependency manager needs better flexibility in general

m-kru, 2019-08-26:
There is one more option. Fetch all dependencies for all tools. Usually the part depending on the tool is small. This approach don't introduce any new option that user has to provide.

olofk, 2019-08-26:
Yes, that would be possible, and also do it for all targets. I'm afraid it might be more complicated with the introduction of custom use flags though. Still, I think the dependency manager needs to have some API eventuall to get all potential dependencies anyway