olofk/fusesoc #301: Documentation regarding VHDL: integration with VUnit, GHDL and yosys

1138-4EB, 2019-08-24:
I use VHDL and GHDL to do CI of several projects; which I can simulate locally with ModelSim and synthesize with Precision to, last, P&R with Vivado. In this context, the advantages provided by VUnit are mainly:

- Dependency analysis and conditional recompilation of VHDL sources.
- Verification Components VHDL library.
- Being able to change the simulator by just setting an environment variable.
- Logging library.

Now, I'd like to extend the build scripts in these projects in order to use fusesoc. The main objective is to make fusesoc consume the (ordered) list of sources from VUnit, and add other SoC elements, to then call different synthesis tools. My target workflows are:

- Synthesis: [ghdlsynth-beta](https://github.com/tgingold/ghdlsynth-beta) (i.e. ghdl + yosys), Precision Synthesis, Vivado.
- P&R: nextpnr, Vivado.
- Targets: ice40hx1k , zynq

I'd like the implementation to be an extension/plugin of the existing entrypoint (the VUnit `run.py`  file). That is, a user should be able to keep using the projects for simulation only, without requiring any modification. Optionally, fusesoc can be installed.

I checked #141 and the blog posts, but I am afraid the content is outdated. It'd be very useful to have a working VHDL example, either in the docs here or in a sibling repo. Nevertheless, I opened this issue to ask about which workflows should I expect to be supported:

| synthesis | P&R | status |
|---|---|---|
| ghdlsynth-beta + yosys | nextpnr | not supported? |
| ghdlsynth-beta + yosys | Vivado | not supported? |
| Precision | nextpnr | not supported? |
| Precision | Vivad) | not supported? |
| Vivado | nextpnr | not supported? |
| Vivado | Vivado | supported |

m-kru, 2019-08-25:
I am not an expert, but I am afraid you will need some script to transform project description from VUnit to FuseSoc. For another way direction this link might be helpful: http://olofkindgren.blogspot.com/2015/12/fusesoc-and-vunit.html

olofk, 2019-08-26:
Thank you for the detailed description of your ideas. There's a bit of overlap between FuseSoC and VUnit and as you have seen in the blog post I have tried to do integration before. When I did that I had to rely on internal FuseSoC API that likely doesn't exist anymore, but the situation today is much better since we now have the EDAM format between FuseSoC and Edalize. Having a well-defined API between FuseSoC and Edalize was created for flows like the one you describe where you already have one half of the solution but need the other half.

So if you already have a list of sources, then it sounds like you don't need FuseSoC's package management features. The way forward would be to create an EDAM file where you describe your sources, parameters, project name and other things and pass it to edalize

This is what the API looks like https://edalize.readthedocs.io/en/latest/edam/api.html and [here](https://gist.github.com/olofk/bd1b178a2ddf279cd84c31e5a14ebe45) is a small example script that can be used to run it with Vivado

I would love to have support for all the flows you mention as well. If you like, you can create issues in edalize for each of them so that we can easier keep track.

1138-4EB, 2019-09-04:
Hi @olofk! Thanks for your reply and my apologies for taking long to respond.

I will do as you suggest. I'll try edalize and will open issues there when I become familiar with the format.

closed 2019-09-04 by 1138-4EB