olofk/fusesoc #303: fusesoc and nyuzi_de2_115 system Error

ecg173gh, 2019-09-19:
Hi! I am trying to build nyuzi_de2_115 system in fusesoc. After solving some errors related with .sv files (changing in .core file the extension of config and define files to svh, and removing the dependency with hardware/fpga/common/synchronizer.sv (not found)), I have the following error that I couldn't solve:

Info (12023): Found entity 1: de2_115_top File: /home/user/or1k/libs/fusesoc-cores/fuse-builds/nyuzi_de2_115_0/src/nyuzi_de2_115_0/rtl/verilog/de2_115_top.sv Line: 19
Error (10932): SystemVerilog error at thread_select_stage.sv(146): can't resolve implicit port connection(s) to instance "scoreboard" without a module declaration or an extern equivalent File: /home/user/or1k/libs/fusesoc-cores/fuse-builds/nyuzi_de2_115_0/src/nyuzi_0/hardware/core/thread_select_stage.sv Line: 146
Error: Quartus Prime Analysis & Synthesis was unsuccessful. 1 error, 1 warning
    Error: Peak virtual memory: 882 megabytes
    Error: Processing ended: Tue Sep 17 12:48:53 2019
    Error: Elapsed time: 00:00:33
    Error: Total CPU time (on all processors): 00:00:23
make: *** [map] Error 3
Traceback (most recent call last):
  File "/usr/local/bin/fusesoc", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/fusesoc/main.py", line 604, in main
    args.func(cm, args)
  File "/usr/local/lib/python2.7/dist-packages/fusesoc/main.py", line 68, in build
    flags, None, args.system, args.backendargs, None)
  File "/usr/local/lib/python2.7/dist-packages/fusesoc/main.py", line 348, in run_backend
    backend.build()
  File "/usr/local/lib/python2.7/dist-packages/edalize/edatool.py", line 117, in build
    self.build_main()
  File "/usr/local/lib/python2.7/dist-packages/edalize/edatool.py", line 126, in build_main
    self._run_tool('make')
  File "/usr/local/lib/python2.7/dist-packages/edalize/edatool.py", line 293, in _run_tool
    except FileNotFoundError:
NameError: global name 'FileNotFoundError' is not defined

Maybe, it is a basic mistake (i am new), but...could someone help me with that?

olofk, 2019-10-23:
Unfortunately it was a very long time since I packaged the Nyuzi core and I haven't looked at it since then. Have you tried adding `hardware/core/scoreboard.sv` to your fileset? It looks like it complains that it can't find a definition of scoreboard. Not sure if that will solve the issue though

imphil, 2020-12-18:
Closing due to inactivity.

closed 2020-12-18 by imphil