olofk/fusesoc #305: error when running 'fusesoc pgm'

rahulkande, 2019-10-07:
Hello, 
I am trying to boot linux on the openrisc with the De0_nano FPGA board. 
I installed the fusesoc by doing: 
1) wget https://github.com/olofk/fusesoc/releases/download/1.9/fusesoc-1.9.tar.gz
2) cd fusesoc
3) pip install --user -e . (I dont have sudo access)
4) fusesoc init
5) fusesoc build de0_nano (I had to remove some FUSESOC_IGNORE files to get it to work)

I connected the de0_nano to my system and ran 'fusesoc pgm de0_nano' but i am getting the following error : 
/home/grads/r/rahulkande/thesis/linux_mor1kx/or_tutorials/tools/fusesoc_v_1.9/fusesoc-1.9/fusesoc/capi2/core.py:611: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  capi2_data = yaml.load(description)
WARNING: ::wb_intercon:0 : plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters
WARNING: ::wb_intercon:1.0 : plusargs section is deprecated and will not be parsed by FuseSoC. Please migrate to parameters
/home/grads/r/rahulkande/thesis/linux_mor1kx/or_tutorials/tools/fusesoc_v_1.9/fusesoc-1.9/fusesoc/capi2/core.py:120: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  _root = Root(yaml.load(open(core_file)))
Traceback (most recent call last):
  File "/home/grads/r/rahulkande/.local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc', 'console_scripts', 'fusesoc')()
  File "/home/grads/r/rahulkande/thesis/linux_mor1kx/or_tutorials/tools/fusesoc_v_1.9/fusesoc-1.9/fusesoc/main.py", line 569, in main
    args.func(cm, args)
  File "/home/grads/r/rahulkande/thesis/linux_mor1kx/or_tutorials/tools/fusesoc_v_1.9/fusesoc-1.9/fusesoc/main.py", line 78, in pgm
    flags, args.system, args.backendargs)
  File "/home/grads/r/rahulkande/thesis/linux_mor1kx/or_tutorials/tools/fusesoc_v_1.9/fusesoc-1.9/fusesoc/main.py", line 310, in run_backend
    backend = get_edatool(tool)(eda_api=edalizer.edalize,
UnboundLocalError: local variable 'edalizer' referenced before assignment

Can someone please help me with this ? 
I have found another issue similar to this but they were getting this error in the build command, I am getting in the pgm command.

olofk, 2019-10-23:
Hi,

Can you try this with a newer version of FuseSoC. Just running `pip install -U --user fusesoc` should get you the latest version (1.9.3) without having to manually download it with wget first.

I should also say that the de0_nano version that is in the orpsoc-cores library is a bit outdated. I would recommend using https://github.com/olofk/de0_nano/ instead. You can add it as a library in your workspace by running `fusesoc library add de0_nano https://github.com/olofk/de0_nano`

rahulkande, 2019-10-26:
Hello, 
I found a workaround for the issue. I just used the sof file generated by building the de0_nano and did the programming directly from the GUI of quartus prime programmer on my windows 10 PC as mentioned here : https://unboxnbeyond.wordpress.com/2013/05/11/configuring-de0-nano-epcs64-flash-device/
The programming was successful this way (I also had to install the drivers for the USB blaster)
Thank you for the reply though

closed 2019-10-26 by rahulkande