olofk/fusesoc #306: --no-export used with local provider results in wrong file path in cache

krishnan-gopal, 2019-10-23:
I have a use-case in which the top-level core file has a local provider, and all its dependencies have remote providers.
I run a modelsim compilation using edalize, and the generated compilation script tells me where the files are located.
If I run without the --no-export option, the compilation script shows that the top-level source file is in my workspace and the source-files of the dependencies are copied from the cache to a src (EXPECTED BEHAVIOR)
However, if I run with the --no-export option, I see the dependencies are used from the .cache directory (EXPECTED), but also the toplevel source-file is listed in the script with a path in the cache directory (WRONG!). The top-level has a local provider and therefore doesnt exist in the cache, so in this case the compilation script specifies a source-file that doesnt really exist.
This must be fixed.

olofk, 2019-10-23:
Hmm.. That sounds very strange. I just tried it here and don't get the same result. Can you run `fusesoc core show <yourcore>` and see if the `CORE ROOT` is set to the expected location?

krishnan-gopal, 2019-10-23:
Just tried, and I am getting the expected location for "Core root" for the toplevel.
One thing to note here is that I dont have version of the toplevel.core file (yet) where the provider is set to the remote source. So, there is no way for it to be checked out into the cache.

olofk, 2019-10-23:
This is super weird and I can't understand at all what's going on. Is this a design you can share with me? At least the core file. Trying to figure out how to debug this

krishnan-gopal, 2019-10-23:
To reproduce this issue, you need a design whose core file exists only with a local provider, and no other version included in your cores root.

Here is a detailed description of my setup.
- I have a toplevel.core file in the path $workspace_toplevel. The provider is set to 'local' here, and all the source-files of this are in the $workspace_toplevel root.
- All the dependencies are coming from a $default_cores-root directory which is specified in my fusesoc config file. The provider is set to remote sources here.
- In my fusesoc config file, I have 2 roots for the core files, as shown:

```
[library.local]
location = .

[library.remote_cores]
location = $default_cores-root
```
- If I run from $workspace_toplevel without the --no-export:

> fusesoc --verbose run --tool=modelsim toplevel

In the path below, I can see my toplevel source-files existing:
> $workspace_toplevel/build/toplevel_0/src/toplevel_0/{PATH TO FILE}/toplevel.vhd

, as well as in the compile script (AS EXPECTED): 
`vcom -2008 -quiet -work testbench {RELATIVE PATH TO FILE}/toplevel.vhd`

- NOW from $workspace_toplevel, If I run with the --no-export:

> fusesoc --verbose run --tool=modelsim --no-export toplevel

In the cache path below, there is NO toplevel source-files existing (EXPECTED, since this has a local provider):
> ../../../../../../../.cache/fusesoc/toplevel_0

, but the compile script specifies this as the path (ISSUE!): 

`vcom -2008 -quiet -work testbench ../../../../../../../.cache/fusesoc/toplevel_0/{PATH TO FILENAME}/toplevel.vhd`

Let me know if you still aren't able to reproduce this. I will build a self-contained example.

krishnan-gopal, 2019-10-23:
OK. I have a very minimalistic example where you can reproduce this issue.
[fusesoc_issue306.zip](https://github.com/olofk/fusesoc/files/3762397/fusesoc_issue306.zip)

Just extract the file, and inside the 'fusesoc_issue306' directory, you will find the required files.
To reproduce the command, just run from that directory:
`~/.local/bin/fusesoc --config fusesoc.conf run --tool=modelsim --no-export toplevel`

Now, look into the `toplevel_0.eda.yml` file (somehow edalize_build_rtl.tcl doesnt contain any files, maybe another issue ?):

```
files:
- file_type: ''
  is_include_file: false
  logical_name: testbench
  name: ../../../../../.cache/fusesoc/toplevel_0/toplevel.v
```
As you can see, the cache directory doesnt contain any toplevel source-files, but fusesoc seems to use this wrong path.

olofk, 2019-10-23:
Just to check, when you say local provider, do you mean that there is no `provider` section at all in the toplevel core file?

krishnan-gopal, 2019-10-23:
No, I mean that under the provider section, the name is set to local

```    
provider:
  name : local
```

krishnan-gopal, 2019-10-23:
Ok, I removed the provider section, and I dont see this issue in the .yml file anymore. However, the generated edalize_build_rtl.tcl still doesnt contain any source-files.

krishnan-gopal, 2019-10-23:
Alright, that sounds good. Thanks for the quick response. We can close this issue as is.

olofk, 2019-10-23:
Cool. Thanks for reporting

closed 2019-10-23 by olofk