olofk/fusesoc #308: FuseSoc with Chisel

Martoni, 2019-11-06:
Is there example of fusesoc usage with chisel ?

olofk, 2019-11-06:
No, there isn't. In theory, this would be a good task for a [generator](https://fusesoc.readthedocs.io/en/master/generators.html) but I'm not enough of a Chisel user myself to know how to best approach it. The closest related example would be the examples I have done of generating verilog from Migen-based cores using FuseSoC generators

Martoni, 2019-11-06:
Ok, thanks for the link. I will look at generator and maybe try to integrate it.

kejace, 2020-01-21:
I'm interested in pursuing something similar for [clash](https://clash-lang.org/). I'm not very familiar with fusesoc yet, but will try to work off of your work @Martoni if you come up with a solution for Chisel.

Martoni, 2020-01-22:
I did not started anything for the moment @kejace . But I'm still interested in ;)

olofk, 2020-01-23:
@kejace @Martoni I think there are two ways to approach this. The first one is to use a generator as I stated above. This would be the preferred option if you want to use FuseSoC for dependency handling and building systems with other FuseSoC cores.

If you already have a dependency manager and a good way to create a verilog file from ClaSH, Chisel or something else, I think that it makes more sense to use [edalize](https://github.com/olofk/edalize) directly. In that case you will take the verilog file (and other files like memory initialization files and constraints) and describe this in an [EDAM](https://github.com/olofk/edalize/blob/master/doc/edam/api.rst) file. Edalize will then consume this file and create the project files for the EDA tool you want to target

kejace, 2020-07-02:
EDAM file generation was recently added to ClaSH via https://github.com/clash-lang/clash-compiler/issues/1386, corresponding PR here: https://github.com/clash-lang/clash-compiler/pull/1410

olofk, 2020-08-18:
That's great. Thanks for letting me know. Hadn't seen this until just now. Hope to see adoption by other projects as well eventually

carlosedp, 2021-04-14:
This has been solved by https://github.com/fusesoc/fusesoc-generators/blob/master/chisel.py.

Check a usage example on: https://github.com/fusesoc/fusesoc-generators/blob/master/template/examples/generators_chisel.core or https://github.com/carlosedp/chisel-playground/blob/master/blinky/chiselblinky.core.

olofk, 2021-04-16:
Ah great! I've been looking forward to closing this for sooooo long. Thanks @carlosedp!

closed 2021-04-16 by olofk

qarlosalberto, 2021-09-01:
@carlosedp @olofk why Chisel support is in FuseSoC and no in Edalize?

carlosedp, 2021-09-01:
Because it's a generator and not a backend. In Fusesoc, we call the chisel-generator that reads the Chisel sources to output the Verilog files that will be processed further down the chain in a similar way as a plain Verilog project while Edalize's job is to generate the backend files to be processed by the vendor toolchain (Symbiflow, Xilinx, etc)

qarlosalberto, 2021-09-01:
It makes sense, thanks! :)

Then I will think how integrate it in TerosHDL without Edalize.