olofk/fusesoc #309: Quartus programming issue

malek-75, 2019-11-07:
In the stable version of fusesoc (sudo  pip install) :  Invoking quartus tool generates an error  
NameError : global name '**remaining**' is not defined.

**remaining** variable is referenced to as an argument in run_main function, quartus.py (edalize) : 
 
This issue doesn't appear in the development version (installed from git)

olofk, 2019-11-07:
I see it now. Technically, the issue is in Edalize rather than FuseSoC but in any case, there is no stable version of Edalize with this fix. I have released version 0.1.7 of Edalize now. Please try `pip install -U edalize` and see if it works after an upgrade

malek-75, 2019-11-07:
Yes it works !
Furthermore, I noticed - in the same file - that a '**board_device_index'** was added to '**the tool_options**'. I guess it concerns boards within a soc as DE1SoC. I will test this and come back to you.
Thanks.

olofk, 2019-11-07:
Yes, that is exactly the reason. See https://github.com/fusesoc/blinky/pull/8 :)

Ok to close this one?

malek-75, 2019-11-08:
Yes.
Thank you.

closed 2019-11-08 by olofk