olofk/fusesoc #310: Fusesoc Imports Python 2.7 Instead of Python3

lchang256, 2019-11-09:
I am working on Ubuntu 14.04 and have Python 3.4.3 installed.  I also set alias to point python to python 3.4.3.  When I start Fusesoc, it gives me the following error:
fusesoc list-cores
Traceback (most recent call last):
  File "/usr/local/bin/fusesoc", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3251, in <module>
    @_call_aside
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3235, in _call_aside
    f(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3264, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 585, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 598, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 791, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (pyparsing 2.0.1 (/usr/lib/python2.7/dist-packages), Requirement.parse('pyparsing>=2.0.2'), set(['packaging']))

I have tried upgrading pyparsing in Python 2.7 but there is no update beyond 2.0.1.  I also searched to see where I could change the pkg_resources import to point to python 3.4.3 but couldn't find it.  Where is that pkg_resources file being read and where can I update it?

Thanks

imphil, 2019-11-19:
fusesoc doesn't support Python 3.4 (only 3.5+); it supports Python 2.7, however. How did you install fusesoc? Avoid "changing" the Python setup that Ubuntu provides; using "pip install fusesoc" will use the default Python 2.7 version, which should work fine (let us know if it doesn't).

olofk, 2020-03-04:
Clsoing due to inactivity

closed 2020-03-04 by olofk