olofk/fusesoc #313: Feature request: Use verilator without building executable

rdaum, 2019-11-15:
In order to facilitate working with existing build systems and external dependencies, it would be nice to be able to have fusesoc kick off verilator and generate sources and the Vxxx.a lib but without attempting to link. This is in fact supposed to be the behavior of Verilator when it is given no C/C++ source files:

> {file.c/.cc/.cpp/.cxx}
> Specifies optional C ++ files to be linked in with the Verilog code. If any C ++ files are specified in this way, Verilator will include a make rule that generates a module executable. Without any C ++ files, Verilator will stop at the module__ALL.a library, and presume you'll continue linking with make rules you write yourself. See also the -CFLAGS option.

However when I provide only verilog filesets to fusesoc, fusesoc + verilator seems to still attempting to linking to an executable and of course get  undefined reference to `main'.  This seems to be because the --exe option is always passed to verilator.  It would be nice to be able to disable the --exe flag, and only get the .a file.  (Also would be nice to be able to specify the destination directory for that .a file)

Basically I want to be able to execute fusesoc from my build system (CMake) to generate sources and the library to link against, but have the build system take over from there to link my own executable, as the build system itself manages all the third party dependencies and so on.

olofk, 2019-11-15:
This sounds like a good use case that I never anticipated. How would you like to proceed with it? I see two options right away based on your comment.
1. Add an option to disable --exe.
2. Disable --exe when there are no c(pp) files.

I think option 1 is better since it's more explicit. In that case, do we also need to avoid passing all c files to the Verilator script file? (Been a long time since I looked at the exact details of this)

As for the output directory I think it's a bit more complicated to set an output directory since the work root is supposed to be decoupled from the directory where you launch FuseSoC. You can however set --build-root and use --system-name to be in bettet control of the work root and name of the created files

rdaum, 2019-11-15:
I think #1 is the better option as well. But the man page does seem to imply that the moment you put them in there that it will add the make rule to build the executable, so it might fight with the no-exe option?

I get the output directory thing; I think the only requirement is that the surrounding build system have a way of getting to the build artifact, just some knowledge of where it will end up. I don't know enough about fusesoc yet to know for sure what that looks like.

olofk, 2019-11-16:
It seems to be safe. I tried a design where I a) remove the --exe flag and b) removed the --exe flag + the c files from the verilator script. I ended up with the same xxx__ALL.a file. Only difference was in VM_USER_CLASSES in the generated makefile.. So this seems safe to do

Re dir: Good, if a known location is the most important part, then it's easy to provide with the options I mentioned above

imphil, 2020-12-18:
Code related to this request is in edalize, and improvements should go there. Please open an issue in edalize or provide a patch to implement this functionality!

closed 2020-12-18 by imphil

jeremybennett, 2021-06-01:
Just to note that I have a use case for this. I'm working on the Open Hardware Group CV32E40P, and we want to use the Verilator model for a number of purposes: smoke test verification, debug server for software engineers and standalone software simulator.  The easiest way is to have a standard library and then let the various environments link to them as they wish.

olofk, 2021-06-02:
Done