olofk/fusesoc #314: Support per-project config files

imphil, 2019-11-18:
It would be nice to have a per-project config file for fusesoc which allows setting default options, like cores root and build directory. (1)

Implementation idea:
Extend the config file search algorithm to go up the directory hierarchy starting from the current working directory until a valid fusesoc.conf file is found. (Current search algo: https://fusesoc.readthedocs.io/en/master/fusesoc.html#core-libraries)

(1) I don't think setting the build directory in the config file is possibly currently, but that's for another issue.

olofk, 2019-11-18:
This is mostly already existing. cores_root is deprecated in favor of the core library sections, but still work if you prefer that. build_root can be set in the config file as well

The only thing lacking is the recursive discovery of a fusesoc.conf but I'm not sure I think that's a good idea as it would be unpredictable to know which file it will pick. You can however specify a file explicitly with --config=path/to/fusesoc.conf

imphil, 2019-11-18:
The "pick a config file up the tree" is reasonably common these days (think git/svn to find their repository root, editorconfig, and many other project config files). 

The `--config` option is less nice for two reasons:
- It's hard to document: you cannot give people a command to copy-paste, instead they need to adjust the path depending on where they are.
- It makes the fusesoc invocation longer again.

I agree that we need to document this behavior well, and that we need to make sure that we give suitable warnings/debug output if conflicts are detected to avoid surprising behavior.

olofk, 2019-11-18:
Oh, I see what you mean now. I thought you meant down the tree. Up the tree makes more sense. There are probably still some things to work out, but let's deal with those when we come across them

imphil, 2019-11-18:
I'm not sure we're having the same definition of up and down in mind :-)

What I mean:
- You call fusesoc in /home/olof/src/greatproject/subdir/subdir2
- fusesoc looks for file named fusesoc.conf in the following directories, and ends the search as soon as soon as it finds the requested file.
  - /home/olof/src/greatproject/subdir/subdir2
  - /home/olof/src/greatproject/subdir
  - /home/olof/src/greatproject
  - /home/olof/src
  - /home/olof
  - /home
  - /

sriyerg, 2019-11-21:
I would assume that any option set on the command line would override the 'default' set in the fusesoc.conf file if/when this is implemented.