olofk/fusesoc #315: Supporting dependency ranges in CAPI file filesets

GCHQDeveloper992, 2019-11-21:
I was thinking it might be beneficial to support a dependency range in CAPI files to cut down on repetitive typing. For example, I have a number of cores that are semantically versioned so I often want to include the range of versions between one major release and the next, e.g.

```yaml
CAPI=2:
name: somevendor:somelibrary:somecore:1.0.0

filesets:
  rtl:
    depend:
      - ">=somevendor:somelibrary:anothercore:1.0.0"
      - "<somevendor:somelibrary:anothercore:2.0.0"
     ....more capi file....
```

For some cores that have lots of dependencies, this gets quite repetitive. I was wondering if the following syntax might be nicer:

```yaml
filesets:
  rtl:
    depend:
      - ">=<somevendor:somelibrary:anothercore:1.0.0/2.0.0"
     ....more capi file....
```

I believe with a small modification to coremanager.py _parse_depend function we could identify the new >=< relation operator, and automatically append two strings to the deps array - one for the >= relation for the 1.0.0 version before the '/' and another for the < relation for the 2.0.0 after the slash.

I could try and put together a quick demo branch of this behaviour to see if that's all that would be involved.

Does that sound like a useful feature for more than just me? Or have I missed a capability that already exists?

GCHQDeveloper992, 2019-11-22:
Having looked at making a quick test branch, I think it might be a bit trickier than I originally thought. Putting in core manager duplicates a bunch of parsing logic from the Vlnv class. I also tried having an @classmethod on Vlnv that could return a list of Vlnvs (so that the Vlnv constructor could be called multiple times) but this causes trouble with handling the return values.

I'll give this another go later on to see if I can find something clearer that achieves the result.

GCHQDeveloper992, 2019-11-22:
I've taken another shot at something that didn't introduce huge numbers of changes. You can see it in my fork [here](https://github.com/GCHQDeveloper992/fusesoc/commit/61f5281946fd3f5475601e64608d2c7ac50fe4f2)

Not sure if this is the neatest implementation though.

olofk, 2020-01-23:
This one slipped between the cracks. Sorry about that. I think the intention is good, but I would suggest using some existing syntax for ranges. The first that comes to mind is the [NPM semver](https://semver.npmjs.com/) which I think is the same that python's own dependency manager uses (or probably something slightly incompatible, just to make life more annoying). Or perhaps that's just more complicated than it needs to be

In the end I guess it's all a trade-off between flexibility and complexity, and we just need to find the right balance.

GCHQDeveloper992, 2020-02-12:
That's interesting, I hadn't found a good example of existing syntax for this sort of thing. I think implementing a couple of the features there should be pretty simple, e.g. the ^, ~ syntax. I'll make some changes to my fork to use ^ and ~ (since that covers the two real uses of the >=< syntax I'd been thinking about). Maybe that'll look OK?

olofk, 2020-03-04:
Closed via #354

closed 2020-03-04 by olofk