olofk/fusesoc #317: Files from Generators can be allocated to a logical_name

a-gibson, 2019-11-25:
When creating a Generator that modifies an existing VHDL file (e.g. set constants to values that relate to build information) I found that the synthesis tool was not able to find the modified VHDL file (it's actually a new file that is based on one from the Git repository).

The other VHDL files in the parent core file are all compiled into a particular library (defined by `logical_name`), and the synthesis tool also expected the modified file to be compiled into the same library (because the original version is).

It seems that this isn't possible with the current definition of `add_files()` in `Generator()`, but should be a straightforward addition.

I should have time later today.

a-gibson, 2019-11-25:
I've created Pull Request #318 for comments.

olofk, 2019-11-26:
I accepted #318 , but I'm pretty sure you could already do this by running add_files([{'file.vhd' : {'file_type' : 'verilogSource', 'logical_name' : 'vhdlSource'}}])
I do realize that the generator helper isn't at all documented so it's not something I expect people will just find out :)

Anyway, feel free to close if one or two of the options works for you

a-gibson, 2019-11-27:
That is very interesting and has taught me a few things, thank you!

Firstly, I didn't actually realise that it was possible to specify `logical_name` per file in a core file. I had seen the examples for `file_type` but hadn't delved into that bit of code so hadn't seen what else was possible. That in itself is useful to know.

Secondly you are absolutely right with your example. `logical_name` can be specified as you suggest but the outcome is subtly different.

If you use:
```
add_files([{'file.vhd' : {'file_type' : 'vhdlSource-2008', 'logical_name' : 'someLib'}}])
```
You get the following in your core file:
```
CAPI=2:
filesets:
  rtl:
    file_type: ''
    files:
    - file.vhd:
        file_type: vhdlSource-2008
        logical_name: someLib
...
```

Whereas, with the patch just merged you can now do this:
```
add_files('file.vhd', file_type='vhdlSource-2008', logical_name='someLib')
```
Which will create the following core file:
```
CAPI=2:
filesets:
  rtl:
    file_type: vhdlSource-2008
    files:
    - file.vhd:
    logical_name: someLib
...
```

So the patch allows a default `logical_name` to be specified in addition to a default `file_type` (which was already supported).

Well, here's hoping that extra flexibility will be useful! As it happens the alternative suggestion would have sufficed for my needs.

closed 2019-11-27 by a-gibson