olofk/fusesoc #322: main.py: yaml.safe_load results in wrong file-name in generated .yml file-list

krishnan-gopal, 2019-12-19:
If the core file contains a file specified with an absolute path (instead of using a relative path to the core-file), like /tools/eda/IP/models/pll/pll.vhd
the generated .yml file contains the file path, with an unknown prefix as shown below: 

```
- core: ::pll:0
  file_type: vhdlSource-2002
  is_include_file: false
  logical_name: lpm
  name: !!python/object/new:fusesoc.capi2.core.String
  - /tools/eda/IP/models/pll/pll.vhd
```
But it should be:
```
- core: ::pll:0
  file_type: vhdlSource-2002
  is_include_file: false
  logical_name: lpm
  name: - /tools/eda/IP/models/pll/pll.vhd
```
The effect of this error can be easily seen if the user runs the run task with the --setup and --run switches seperately, in which case the --run uses the generated .yml file from --setup.

The solution is to use a YAML fullloader in the main.py as shown:

Currently:
```
edam = yaml.safe_load(open(eda_api_file))
```

Solution:
```
edam = yaml.load(open(eda_api_file), Loader=yaml.FullLoader)
```

olofk, 2019-12-20:
I was hoping that the test suite would catch these kind of issues :(

I fixed it another way by just casting the file name to string. Can you check if the latest commit fixes the issue. Maybe FullLoader is a better way but I don't have the opportunity to check if it would cause other errors right now

krishnan-gopal, 2020-01-02:
Ok, it works now with the latest commit. Thanks.

closed 2020-01-02 by krishnan-gopal