olofk/fusesoc #329: Windows: mixed directory separators in fusesoc init

imphil, 2019-12-27:
I get the following output in CI: 

```
fusesoc init -y
INFO: Writing configuration file to 'C:\Users\appveyor\.config\fusesoc\fusesoc.conf'
INFO: Initializing orpsoc-cores
INFO: Cloning library into C:\Users\appveyor\.local/share\fusesoc\orpsoc-cores
Cloning into 'C:\Users\appveyor\.local/share\fusesoc\orpsoc-cores'...
INFO: Initializing fusesoc-cores
INFO: Cloning library into C:\Users\appveyor\.local/share\fusesoc\fusesoc-cores
Cloning into 'C:\Users\appveyor\.local/share\fusesoc\fusesoc-cores'...
INFO: FuseSoC is ready to use!
fusesoc list-cores
```

Note the mix of `/` and `\` in the path. We should have a closer look at that and ensure that we use Python functions which build a valid path for each OS.

I didn't notice any bugs due to this, so for now it's purely cosmetic.

imphil, 2020-12-18:
Likely to be resolved as part of #376

olofk, 2025-02-21:
fusesoc init was removed, so closing this