olofk/fusesoc #331: Enable PR builds for readthedocs

imphil, 2020-01-06:
ReadTheDocs has an experimental feature to enable pull-request builds. To get added to their beta the repo owner needs to mail them, instructions are here: https://blog.readthedocs.com/building-docs-for-pull-requests/ 

This is actually quite nice to see doc changes rendered in a PR, see https://github.com/cocotb/cocotb/pull/1282 for an example (look at the " continuous-documentation/read-the-docs" check and click on "Details")

@olofk can you enable that feature?

olofk, 2020-01-06:
Mail sent. Will keep you posted

olofk, 2020-01-09:
This should be done now. Not sure how to test it. Can you let me know if it works as expected?

imphil, 2020-01-09:
It should be visible once a new PR is opened, I'll check that on the next one.

imphil, 2020-01-09:
No, doesn't work yet. I think I had to remove the RTD webhook in the GitHub project and then re-add it through the RTD admin interface when I did the same thing for cocotb.

olofk, 2020-01-13:
Done now. Thanks for iniating this

closed 2020-01-13 by olofk