olofk/fusesoc #332: attr() got an unexpected keyword argument 'converter'

MBratic, 2020-01-06:
Hello,

after "pip install fusesoc" is finished "succesfully" I run i.e. the command "fusesoc list-cores", I get the following error message.

Traceback (most recent call last):
  File "/net/home/bratic/.local/bin/fusesoc", line 11, in <module>
    load_entry_point('fusesoc', 'console_scripts', 'fusesoc')()
  File "/net/home/bratic/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/net/home/bratic/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2843, in load_entry_point
    return ep.load()
  File "/net/home/bratic/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2434, in load
    return self.resolve()
  File "/net/home/bratic/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2440, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/net/heap/bratic/risc-v/tools/01_src/fusesoc/fusesoc/main.py", line 16, in <module>
    from fusesoc.coremanager import CoreManager, DependencyError
  File "/net/heap/bratic/risc-v/tools/01_src/fusesoc/fusesoc/coremanager.py", line 6, in <module>
    from simplesat.constraints import PrettyPackageStringParser, Requirement
  File "/net/home/bratic/.local/lib/python2.7/site-packages/simplesat/__init__.py", line 3, in <module>
    from .constraints import Requirement, InstallRequirement
  File "/net/home/bratic/.local/lib/python2.7/site-packages/simplesat/constraints/__init__.py", line 5, in <module>
    from .constraint_modifiers import (
  File "/net/home/bratic/.local/lib/python2.7/site-packages/simplesat/constraints/constraint_modifiers.py", line 85, in <module>
    class ConstraintModifiers(object):
  File "/net/home/bratic/.local/lib/python2.7/site-packages/simplesat/constraints/constraint_modifiers.py", line 86, in ConstraintModifiers
    allow_newer = attr(**_coerced_set)
TypeError: attr() got an unexpected keyword argument 'converter'

Are there other versions of the packages nedded?

Best regards

olofk, 2020-01-06:
I have not seen this before and it looks like it comes from one of the FuseSoC dependencies. Could you try to run `pip3 install fusesoc` instead? I notice that your installation uses python 2 so it could ve that one of the dependencies has a python 2 problem.

MBratic, 2020-01-09:
After I use pip3 to install fusesoc it works out! Thank you very much!

olofk, 2020-01-09:
Thanks for letting me know. Perhaps it's time to say goodbye to python2 support for fusesoc now

imphil, 2020-01-10:
We do have CI for Python 2.7 and things work there. So let's have a closer look at the versions you're using. Can you give us the output of the following commands:

```
$ pip show attrs
$ pip show simplesat
$ pip show fusesoc
```

imphil, 2020-01-15:
Closing this for now, please reopen if you'd like to provide more input.

closed 2020-01-15 by imphil