olofk/fusesoc #334: Decide on, document, and enforce coding style

imphil, 2020-01-06:
It would be nice to decide on, document, and enforce a common coding style. 

The base: PEP8 (https://www.python.org/dev/peps/pep-0008). Most other style guides build on top of this base style guide and are compatible with it. Since PEP8 leaves many freedoms, tools have evolved on top of it.

Since nobody wants to (re-)format large pieces of code, the style guide discussion can be mostly reduced to something that is produced by a code formatting tool. For Python, we do have a couple options. Some of these tools are very relaxed and leave the code author a bit of stylistic freedoms (at the expense of possibly arguing with a reviewer about the right way of doing things), others forcefully reformat the code to match the one-and-only-right-formatting (at the expense of taking away stylistic freedom).

Well-known options, in the order of least opinionated to most opinionated:
- autopep8: https://pypi.org/project/autopep8/ Reformats code which doesn't match PEP8. Rather configurable, and widely used.
- Yapf: Reasonably configurable, comes with a couple profiles for existing coding styles.
- Black: Very minimal set of configuration, enforces its own coding style (https://black.readthedocs.io/en/stable/the_black_code_style.html)

My view:
- Yapf (together with isort) supports the Google Coding Style, which is one of the most detailed coding styles. So linking to http://google.github.io/styleguide/pyguide.html is a good way to document our coding style.
- We use that in OpenTitan and we found that it produces unstable output: the output differs between Python versions and yapf versions. [Others seem to have similar issues](https://news.ycombinator.com/item?id=17155205).
- I didn't use black before, but I've heard reasonably good things about it. The lack of a detailed style guide in written form is more than outweighed by the instruction "simply call black and don't worry".

I've created https://github.com/olofk/fusesoc/pull/333 as a proof-of-concept PR to show how a tree reformatted with black could look like.

olofk, 2020-01-06:
I'm fine with this. Looking at the changes it's mostly line breaks and changing single to double quotations. Should we hook it up to CI as well to have pre/post-commit checks?

imphil, 2020-01-06:
Yes, CI is essential. I'll follow up with a PR for that.

imphil, 2020-01-10:
Fixed in #333

closed 2020-01-10 by imphil