olofk/fusesoc #337: Drop Python 2-support

imphil, 2020-01-10:
According to the `pip install -U pypistats --user && pypistats python_minor fusesoc`` we do have .55% of our users on Python 2.7.

```
$ pypistats python_minor fusesoc
| category | percent | downloads |
|----------|--------:|----------:|
| 3.5      |  75.12% |    37,041 |
| 3.7      |  22.32% |    11,005 |
| null     |   0.97% |       480 |
| 3.6      |   0.92% |       455 |
| 2.7      |   0.55% |       269 |
| 3.8      |   0.06% |        32 |
| 3.4      |   0.05% |        25 |
| 2.6      |   0.01% |         4 |
| Total    |         |    49,311 |

Date range: 2019-07-13 - 2020-01-08
```

How do people feel about removing Python 2 support? (Old releases are still available with Python 2 support, of course.)

wallento, 2020-01-10:
Yeah, I like that idea and am all for it. The only concern I would have is that the target industry is absurd when it comes to ancient tool versions. The stats are clear, so lets go for it.

imphil, 2020-01-10:
A bit of research shows:

Ubuntu 16.04: Python 2.7.12 and 3.5.2
Ubuntu 18.04: Python 2.7.15 and Python 3.6.7
RHEL/CentOS 6: Python 2.6.6 (no Python 3)
RHEL/CentOS 7: Python 2.7.5 and Python 3.6.8

So: RHEL6 is already not supported without a [SCL](https://www.softwarecollections.org/en/) (which are rather easy to use). RHEL7 has a recent-enough version of Python 3.

(Slightly more painful is the support for Python 3.5 in Ubuntu 16.04, but that will stay with us for a bit longer.)

olofk, 2020-01-10:
Thanks for this. It's specifically the RHEL/CentOS users I've been most worried about but it looks like supporting 2.7 won't really to any difference for them. Come to think of it, how many of those 269 py2 downloads are from Travis/Appveyor? :)

Why do you think it's painful with py3.5? From my perspective I think it's perfectly fine to have that as a minmum version but perhaps I'm missing something

I'm ready to drop py2

wallento, 2020-01-10:
IMO type annotations have been added after 3.5 which are really useful for a clear codebase

wallento, 2020-01-10:
https://docs.python.org/3/whatsnew/3.6.html

olofk, 2020-01-10:
Agree that it would be nice but doesn't outweigh losing 3.5 support at this point

Jbalkind, 2020-01-10:
I personally don't think this should be a priority. The additional effort at present is not that significant, is it?

imphil, 2020-01-10:
@Jbalkind do you have use/data of use for fusesoc and Python 2?

wallento, 2020-01-10:
I think the differentiation is if it means someone goes to remove every compatibility stuff or if just nothing new is added. The latter should happen IMO, the former is indeed something not at highest priority, but those compatibility things could be removed because adding non-compliant code from now on renders them useless..

So an imminent action list is to advertise it as Python 3 only and don't accept any new compatibility code. Cleaning code up is a general task for FuseSoC, that @imphil tackles right now if I read his PRs properly ;)

imphil, 2020-03-04:
OK, thanks everyone for their input. I now have code cleanups and new features ready that will depend on Python 3. Can we:

- Make a last release that supports Python 2? If people want to continue using Python 2, they are welcome to stay on that release, it will continue to work as usual.
- Once that's done, I'll do some cleanups as I move along through the code base.

olofk, 2020-03-04:
I'm about to release a new FuseSoC version now. Let this be the last one with guaranteed py2.7 compatibility. Shout now if there's anything you need to have in this release

PR #369 (2020-03-10): Remove support for Python 2.7
Python 2.7 reached EOL in the beginning of this year after being
supported for roughly 10 years. It's now time also for fusesoc to
benefit from the language and ecosystem improvements that happened since
then, and drop backwards-compat with Python 2.7. The last version to
support Python 2.7 is fusesoc 1.10, which can still be installed through
pip.

Fixes #337
commit 615dcbf1 (2020-03-10): Remove support for Python 2.7

Python 2.7 reached EOL in the beginning of this year after being
supported for roughly 10 years. It's now time also for fusesoc to
benefit from the language and ecosystem improvements that happened since
then, and drop backwards-compat with Python 2.7. The last version to
support Python 2.7 is fusesoc 1.10, which can still be installed through
pip.

Fixes #337
closed 2020-03-10 by olofk