olofk/fusesoc #341: How to pass compiler directives in core file

rahulraveendran15-coder, 2020-01-22:
Hi,
How to pass compiler directives in .core files in fusesoc. 

Consider an example:

´ifdef synth
`ifdef cntrl
a = b
´else
a = c
`else
a = d
`endif

how to pass the above define "synth" in .core file (along with some file).

olofk, 2020-01-23:
Hi,

You do this in two steps. First you create a parameter in the parameters section (with an optional default value)
```yaml
parameters:
  synth:
    datatype: bool
    default : true #This is optional and can be set or overridden in the target or on the command line
    paramtype: vlogdefine
```

In your targets you can now use this parameter and optionally set a default value for it. Writing down a few 
```yaml
targets:
  some_synth_target:
    parameters : [synth] #This will inherit the value of the synth parameter from the parameters section
   some_other_synth_target:
    parameters: [synth=true] #This will force the define to be set for this target
  some_sim_target:
    #If there is no parameters key in the target, the value of the synth define will be false
  some_other_sim_target:
    parameters : [synth=false] #If you have set synth : true in the parameters section, you can force it to be false for this target

For the example you used, I guess you always want it set on synthesis targets and false otherwise, but if you for example have a define that you might want to change between runs, let's call it `mydefine`, then you can specify it without a value in the target and then set it on the command-line instead like this

`fusesoc run --target=some_target some_core --mydefine=true`

If you run `fusesoc run --target=some_target some_core --help` you will get a list of all parameters defined for a target. Parameter types can be vlogdefine, vlogparam, generic, plusarg or cmdlinearg. Not all backends handle all parameter types. For example, only backends that understand VHDL handle generics and only backends that support verilog handles vlogdefine, vlogparam and plusarg.

Hope this made it a bit clearer. Let me know otherwise

rahulraveendran15-coder, 2020-01-24:
Hi,

I have tried the method you suggested but the defines are not passing as what I am giving. Let me show you my requirement in detail below:

`ifdef ASIC
 `ifdef SYNTH
  assign a =b;
`else
assign a = c;
`endif
`else
assign a = d;
`endif

I am using systemverilog files and I tried the method you suggested with  vlogdefine, vlogparam and plusarg

This is how my core file looks like:

CAPI=2:
name : ::clkgate:1-r1
filesets:
  rtl_files:
    files:
      - check.sv: {is_include_file : true}
      -rtl.sv
    file_type : systemVerilogSource

  tb_files:
    files:
      - tb.sv
    file_type : systemVerilogSource

parameters:
  ASIC:
    datatype: bool
    default : true #This is optional and can be set or overridden in the target or on the command line
    paramtype: plusarg

targets:
  default:
    filesets : [rtl_files]
  sim:
    default_tool : modelsim
    filesets : [rtl_files, tb_files]
    toplevel : tb
    parameters : [ASIC] 

Can you please help on this

imphil, 2020-01-24:
Your core file looks OK. Please check how modelsim is called by running `fusesoc --verbose`, and look at the generated files in the `build` directory.

To post code on GitHub issues, you can use fenced code blocks with three backticks as described here: https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks#fenced-code-blocks This ensures that indentation is retained, which is important for YAML files.

rahulraveendran15-coder, 2020-01-24:
Sorry I am new to fusesoc
what is meant by how modelsim is called?

imphil, 2020-01-24:
After fusesoc completes its work, it creates a Makefile. This Makefile can be found in the `build` folder, and contains instructions how to call ModelSim. Please have a look at this Makefile to see how ModelSim is invoked. If you cannot figure it out, please zip the build folder and attach it to this bug. (Ensure that you don't have any sensitive information in it.)

rahulraveendran15-coder, 2020-01-24:
Hi ,
Thanks for the support. I deleted and typed the content again the parameter section and right now its working fine as expected.

imphil, 2020-01-24:
Great!

closed 2020-01-24 by imphil

rahulraveendran15-coder, 2020-01-24:
how to pass the or override defines in commands line in the below example

CAPI=2:
name : ::test:1-r1
filesets:
rtl_files:
files:
- check.sv: {is_include_file : true}
-rtl.sv
file_type : systemVerilogSource

tb_files:
files:
- tb.sv
file_type : systemVerilogSource

parameters:
ASIC:
datatype: bool
default : true #This is optional and can be set or overridden in the target or on the command line
paramtype: vlogdefine
 SYNTH:
    datatype: bool
    default : false #This is optional and can be set or overridden in the target or on the command line
    paramtype: vlogdefine

targets:
default:
filesets : [rtl_files]
sim:
default_tool : modelsim
filesets : [rtl_files, tb_files]
toplevel : tb
parameters : [ASIC,SYNTH]

my run command is "fusesoc run --target=sim --tool=modelsim" <core_name> --SYNTH=true"
When runnign the above command it is throwing error  "error: argument --SYNTH: ignored explicit argument 'true'"

How to solve this?