olofk/fusesoc #342: CentOS/RHEL 7 Git Doesn't Support Patching Option

GCHQDeveloper560, 2020-01-23:
The `git apply` command used in the provider module as a better patch command uses the `--unsafe-paths` option. I've not found when git added this option, but it's not supported in git 1.8.3 shipped with RHEL/CentOS 7. Is there another patch command that can be used, or is a required git version documented somewhere?

imphil, 2020-01-23:
No, we don't document a required git version at this point. That would be good to do, and also figuring out a solution which works on RHEL7. Do you have a command that works instead of the `git apply` one?

olofk, 2020-01-23:
I have considered many times to use a python-based patch tool instead to avoid relying on external commands, and perhaps this would be a good opportunity for that. The thing that has held me back in the past is that the patch library I found last time (some years ago) by design only allowed file changes and couldn't create/remove files. A quick googling gives the impression that the situation is better now

GCHQDeveloper560, 2020-01-24:
It looks like this option was added in git 2.3.3. We could check the git version when constructing the argument list. The Python option might be best if you have found a better module, but I assume we'll need to add several tests to check that it supports the patch operations we need.

GCHQDeveloper560, 2020-01-24:
Commit 9e016c9 is a is a version checking workaround that seems to work for me. If you'd like to stick with git I can submit a PR.

olofk, 2020-01-27:
That looks like a good enough solution to make it work on RHEL7. Please submit a PR. Just to make sure, does earlier git behave like newer git with --unsafe-paths set? I honestly don't remember why it was done like this in the first place, but I'm sure I had a good reason :)

GCHQDeveloper560, 2020-01-28:
I believe you're right that older git behaves like newer git with `--unsafe-paths` set. My understanding is 2.3.3 introduced a new check that wasn't there before and is disabled with `--unsafe-paths`.

I did a little more investigation with a newer git and removing the `--unsafe-paths` option, which causes the `test_github_provider` item in `test_provider.py` to fail. The command being run is

```
git apply --directory <tmpdir>/vlog_tb_utils_1.1 <path to repo>/fusesoc/tests/cores/vlog_tb_utils/files/0001-testpatch.patch
error: invalid path '<tmpdir>/vlog_tb_utils_1.1/vlog_functions.v'
```

The patch isn't doing anything odd that should trigger a check. It appears that `--directory` doesn't change git's working directory, it just prepends the directory to paths in the patch. Therefore, `--directory` seems useless without `--unsafe-paths`, which seems silly. If I `cd` to the directory specified by `--directory` and run the git command without any options it works.

So, it seems the options would be
1. Continue using `--directory` and disable the check with `--unsafe-paths` in newer versions of git as in the commit.
2. Run git from the directory where the patch is being applied rather than using `--directory`

Unless I'm missing something the second option is simpler and doesn't require disabling the check. `Launcher` seems to support specifying a working directory, so the test passes with the following:
```
                    Launcher(
                        "git",
                        [
                            "apply",
                            patch_file,
                        ],
                        self.files_root,
                    ).run()
```

GCHQDeveloper560, 2020-01-28:
After a few more experiments it seems `--directory` isn't totally useless without `--unsafe-paths`. You can use it with relative paths below the curent working directory, so something like `cd /tmp` followed by `git apply --directory <modified relative path>` without `--unsafe-paths` works.

I've submitted PR #346 with the second approach.

olofk, 2020-01-28:
Thank you for the careful investigation. The approach you chose definitely feels like the right approach and simplifies things. Patch is squashed and merged

closed 2020-01-28 by olofk