olofk/fusesoc #344: How to override or pass defines from command line

rahulraveendran15-coder, 2020-01-24:
how to pass the or override defines in commands line in the below example

CAPI=2:
name : ::test:1-r1
filesets:
rtl_files:
files:

check.sv: {is_include_file : true}
-rtl.sv
file_type : systemVerilogSource
tb_files:
files:

tb.sv
file_type : systemVerilogSource
parameters:
ASIC:
datatype: bool
default : true #This is optional and can be set or overridden in the target or on the command line
paramtype: vlogdefine
SYNTH:
datatype: bool
default : false #This is optional and can be set or overridden in the target or on the command line
paramtype: vlogdefine

targets:
default:
filesets : [rtl_files]
sim:
default_tool : modelsim
filesets : [rtl_files, tb_files]
toplevel : tb
parameters : [ASIC,SYNTH]

my run command is "fusesoc run --target=sim --tool=modelsim" <core_name> --SYNTH=true"
When runnign the above command it is throwing error "error: argument --SYNTH: ignored explicit argument 'true'"

How to solve this?

olofk, 2020-01-27:
Ah.. It took me a while to figure that out. Bool parameters are a bit special. You can't set a value, but only force them to be true. The correct way to force it to be true is to run 

    fusesoc run --target=sim --tool=modelsim" <core_name> --SYNTH

This also means that there is a limitation with bool parameters in FuseSoC. You can't force them to be false on the command-line. If that's ok in your case you can keep using a bool parameter. Otherwise you might want to change it to `datatype : int` and use 0 and 1 for false/true values

olofk, 2020-01-27:
I think it would be good to do like you tried to do, so I created a bug report for this, so that we don't forget about it https://github.com/olofk/fusesoc/issues/345

rahulraveendran15-coder, 2020-01-28:
actually i found one more problem if I am declaring the parameters like this 
parameters:
  ASIC:
    datatype: bool
    default : false #This is optional and can be set or overridden in the target or on the command line
    paramtype: vlogdefine

and in target section if I am doing  like this 
targets:
  default:
    filesets : [rtl_files]
  asicsim:
    default_tool : modelsim
    filesets : [rtl_files, tb_files]
    tools:
      modelsim:
        vlog_options: [-timescale=1ns/1ns]
    parameters : [ASIC,SYNTH] 

 Consider I want to run FPGA target. But while running fusesoc it is not taking the ASIC define as false,  since both ASIC and SYNTH I defines as false in parameter. It is taking ASIC target.

olofk, 2020-01-28:
Ah, I think I see whay you mean now. This should be fixed, but until I find a good solution I think the best way in your case is to not set a default value in the parameters section.

For targets where you always want for example ASIC to be true, you can set it like `[ASIC=true,SYNTH]'. For targets where you want to switch between true and false you can just set `[ASIC,SYNTH]` and then use `--ASIC` on the command-line to set it true.

imphil, 2020-12-18:
The solution that @olofk presented should be sufficient; there's more to be done in parameter handling and overrides, but this issues isn't the best place to take actionable items from. Let's close it for now.

closed 2020-12-18 by imphil