olofk/fusesoc #347: Multi dependencies on core

rahulraveendran15-coder, 2020-01-30:
How to use multi dependencies in core?

CAPI=2:
name : ::test:1-r1
filesets:
rtl_files:
files:

check.sv: {is_include_file : true}
 depend:
      - wiredelay
-rtl.sv
file_type : systemVerilogSource
tb_files:
files:
tb.sv
file_type : systemVerilogSource
parameters:
ASIC:
datatype: bool
default : true #This is optional and can be set or overridden in the target or on the command line
paramtype: vlogdefine
SYNTH:
datatype: bool
default : false #This is optional and can be set or overridden in the target or on the command line
paramtype: vlogdefine

targets:
default:
filesets : [rtl_files]
sim:
default_tool : modelsim
filesets : [rtl_files, tb_files]
toplevel : tb
parameters : [ASIC,SYNTH]

Core file of wiredelay:
CAPI=2:
name : ::wiredelay:1-r1
filesets:
rtl_files:
files:
sample.sv: {is_include_file : true}
-sample2.sv
file_type : systemVerilogSource
tb_files:
files:
tb1.sv
file_type : systemVerilogSource
parameters:
ASIC:
datatype: bool
default : true #This is optional and can be set or overridden in the target or on the command line
paramtype: vlogdefine
SYNTH:
datatype: bool
default : false #This is optional and can be set or overridden in the target or on the command line
paramtype: vlogdefine

targets:
default:
filesets : [rtl_files]
sim:
default_tool : modelsim
filesets : [rtl_files, tb_files]
toplevel : tb
parameters : [ASIC,SYNTH]

My aim is to use this wiredelay core in test core by taking the defines from wiredelay.
That means I need to use target sim from wiredelay core and the same defines that passing the target sim of wiredelay needs to be applicable for test core.

Could you please help me on this

That means when i run fusesoc --target=sim test  to which core this target sim is applicable? test or wiredelay
which defines will go through these two cores?

olofk, 2020-01-30:
When a core is used as a dependency of another core (like wiredelay is a dependency of test), the `default` target will always be used in the dependencies. FuseSoC will pick up the filesets, parameters, tool options and other things (but not toplevel). So if you have a parameter defined in wiredelay, it will be visible when you run the test core that has a dependency on wiredelay

rahulraveendran15-coder, 2020-01-30:
So the default target from wiredelay is taken 
do the parameters declared in wiredelay for eg ASIC and SYNTh do I need to declare again in test core file ? Or Can i use parameters directly in test cores target without declaring as parameter in test core file
Some where I read in document that we need to make scope as public inorder to get parameter visibility in all  cores in case of dependency

rahulraveendran15-coder, 2020-01-31:
Also one  more clarification needed. I will explain my requirement as below.
My test core is dependent on wiredelay as above. consider I used parameters in default target as ASIC only. But while running test core i have given target as sim which contains both ASIC and SYNTH.
So how to give this SYNTH parameter to wiredelay core also since it is not defined in default target? Similarly if I am using some other parameters or defines

olofk, 2020-01-31:
The parameters that are declared in the default target in wiredelay will be visible by test. The `public` option was used for the old core description format (CAPI1) but is not used in CAPI2 files.

Unfortunately I realize there is another bug in the parameter handling. It does not work to declare a parameter (e.g. ASIC) in the default target of wiredelay, and set a value (e.g. `parameters : [ASIC=true]`) in the test core. FuseSoC will complain that it can't find the ASIC parameter in that case. This should be fixed, but until then there are two workaround
1. Set `--ASIC` on the command-line. Then you only need to declare the parameter in the wiredelay core.
2. Declare the ASIC parameter in the test core too and use it in the sim target. In this case FuseSoC will take use that parameter and use the one from wiredelay if it's not declared. Perhaps this is better shown as the example below

wiredelay.core
```yaml
...
targets:
  default:
    ...
    parameters : [ASIC]

parameters:
  ASIC:
    ...
```

test.core
```yaml
...
targets:
  sim: #When using this target, FuseSoC will use the ASIC parameter from the test core
    ...
   parameters : [ASIC]

  other_target: #When using this target, FuseSoC will use the ASIC parameter from wiredelay
    ...

parameters:
  ASIC:
    ...
```

rahulraveendran15-coder, 2020-02-02:
I need to take target from wiredelay all the time. But I have multiple parameters in wiredelay such as [ASIC,FPGA,SYNTH]. these parameters I need to use true and false like different combinations. since only default target is taking from wiredelay in that target only one value or one combination values for these parameters will be there. Is this thing possible?

imphil, 2020-12-18:
I'm not sure what actionable items are left in this issue. @rahulraveendran15-coder please have a look at the revamped documentation to check if that answers your questions, and open an individual issue for things which are not possible currently.

closed 2020-12-18 by imphil