olofk/fusesoc #348: pip hangs on install fusesoc

ThomasHornschuh, 2020-01-30:
Hi. I'm just setting up a new development VM based on Ubunto 18.04LTS.
Trying to install fusesoc (which I actually have done dozens of times before) hangs.
I tried direct install with pip
    `sudo pip(3) install fusesoc`

and also the  git clone variant.
`sudo  pip3 -vvv install -e .`

gives
````
Obtaining file:///home/thomas/development/fusesoc
  Running setup.py (path:/home/thomas/development/fusesoc/setup.py) egg_info for package from file:///home/thomas/development/fusesoc
    Running command python setup.py egg_info
````

At this point it hangs, control-c gives
````
^CCleaning up...
Operation cancelled by user
Exception information:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 342, in run
    requirement_set.prepare_files(finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 518, in _prepare_file
    abstract_dist.prep_for_dist()
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 129, in prep_for_dist
    self.req_to_install.run_egg_info()
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 439, in run_egg_info
    command_desc='python setup.py egg_info')
  File "/usr/lib/python3/dist-packages/pip/utils/__init__.py", line 694, in call_subprocess
    line = console_to_str(proc.stdout.readline())
````
pip instead of pip3 gives the same result. Any idea?
It may not be fusesocs fault, can also be something with python....

olofk, 2020-01-31:
I have not seen this before so I would like to point the blame to Python/pip for now as I have no idea how to reproduce it or fix it. Hope it works out and I will let you know if I hear anything similar

ThomasHornschuh, 2020-01-31:
I'm investigating it, and keep you posted. Some research shows, that most likely the problem occurs because I tried to install python packages with pip3 and pip on the same machine. Actually I'm building a new development VM because I like to update all my working environment to newer versions. Among this also migrating to python 3.x and updating fuseseoc.
The python 3 vs 2 issue in general is a mess. Sorry, 'm just ranting :-)
Maybe at the end I find some conclusions and advice to be added to FuseSoC documentation.

ThomasHornschuh, 2020-02-01:
We can close it, the problem has disappeared, most likely because the machine was rebooted in the meantime.
There are things in the world which will never change.

closed 2020-02-01 by ThomasHornschuh